freifunk-gluon/gluon #591: Problem in gluon-neighbour-info?

AKA-47, 2015-12-08:
Seems there is a problem in gluon-neighbour-info:

New statuspage shows

Mesh-VPN
    true
Automatische Updates
    false
Branch
    stable

but

uci show | grep vpn

simple-tc.mesh_vpn.enabled='0'
simple-tc.mesh_vpn.ifname='mesh-vpn'

vs

gluon-neighbour-info -d ::1 -p 1001 -t 1 -c 1 -r nodeinfo

fastd":{"enabled":true,"version":"v17"},
"autoupdater":{"enabled":false,"branch":"stable"}

but 

uci show

autoupdater.settings=autoupdater
autoupdater.settings.version_file='/lib/gluon/release'
autoupdater.settings.enabled='0'
autoupdater.settings.branch='experimental'
autoupdater.experimental=branch

NeoRaider, 2015-12-08:
I've found the issue in the fastd announce script, but I can't find anything wrong with the autoupdater one.

NeoRaider, 2015-12-08:
The fastd issue has been fixed by 5ecb8a73668d030561dda8cf22143bedb27b05da.

Did you by any change change the autoupdater branch via SSH? At the moment, UCI configs aren't reloaded by gluon-announced, so a reboot is necessary to see the changed values.

AKA-47, 2015-12-11:
Found the problem with autoupdater .. Reboot did it finally :D

closed 2015-12-11 by AKA-47