freifunk-gluon/gluon #592: adding possibility to reduce bandwith consumption of a FF node

Tarnatos, 2015-12-09:
After several german isp had dicided to reduce the internet connection speed after reaching a monthly bandwith limit some people asked me for an option in gluon to reduce bandwith consumption.

This could be realized with:
- disable VPN connection in an adjustable period of time (11pm till 7am for example) to save broadcast traffic
- possibility to change the amount of fastd gateway connections
- disable the uplink connection after reaching an predefined amount of monthly traffic and disable the client network if no internet connection (uplink or mesh) is avaliable, should turn back on if some mesh partner with active internet connection is avaliable

Most of this requested features are already avaliable yet by using ssh cmds. For untrained "normal Freifunk consumers" this feature(s) should be included in the config site so they could be turned on during installation process.

rotanid, 2015-12-09:
although these options, especially the first one, would be nice - i'm interested which ISP's limit is so low, a FF-node would consume all the traffic? none of the big ones....

Tarnatos, 2015-12-09:
o2 for example limits consumer dsl connection contracts after reaching 300GB (16MBits contract) or 500GB (50MBits or more) during a billing period for the rest of the period to 2 or 8MBits.

I dont want to say that a FF node consumes all this traffic by itself but if your dsl connection is alredy heavyly used by your family members (YT Netflix Steam), 300GB is reached quickly. The owner will then study the traffic charts and will notice that the FF node produces a lot of traffic too. In order to keep the monthly costs low the FF node will be disconnected and will not come back again.

Futhermore in big domains FF nodes produces even 10 till 30GBs (sometimes more) off broadcast traffic without having one client.

This topic is discussed in the Freifunk Forum too:
https://forum.freifunk.net/t/freifunk-an-internet-anschluessen-mit-drosselung/9634

jplitza, 2015-12-09:
Won't happen, see #486.

closed 2015-12-09 by jplitza