freifunk-gluon/gluon #594: old Statuspage

fragstone, 2015-12-09:
Hi,

ok the new Statuspage has some good features and is a eyecatcher but is it possible to make a gluon package availabe with the old statuspage?
It will be nice when people are able to choice the right package for yourself?

regards
Thomas

pinguinpfleger, 2015-12-09:
Why do you want to go back to the old status page?

fragstone, 2015-12-09:
sorry pinguinpfleger :-)

but that is not the question. 
it are plus points for the new one and plus points for the old one.
so i think over the next month users will use the new one. 

but actual i think users shout be able th choise it.

mmalte, 2015-12-09:
Some possible issues:

The load rises significantly when someone is viewing the the status page for some minutes.  
Only one (or a few) user at a time is able to use the status page. 
The contact information is displayed.

pinguinpfleger, 2015-12-09:
I just wanted to here your intention why you seem to prefer the old one, no big deal.

fragstone, 2015-12-09:
one big thing for me is that i can check the status page by linky (lynx) [text console browser]

pinguinpfleger, 2015-12-09:
mmalte: if that's true the new status page is a serious vulnerability for DOS attacks :(
maybe we could improve the status page by splitting it up in specific sub status pages and not generate all status data at once.

mmalte, 2015-12-09:
In a huge Mesh with 960 online nodes I saw a 841v10 reboot when I visit the status page. 

Had no time to reproduce the issue.

tcatm, 2015-12-09:
@fragstone Sure, it may still be possible to use the old status page and there could even be a package if someone volunteers to maintain it.

fragstone, 2015-12-09:
Hi Tcatm,

stupid question :D who i can found the old code to see if i can maintenance it.

jplitza, 2015-12-10:
@fragstone [This](https://github.com/freifunk-gluon/gluon/tree/431ac81407919f9e2a88ea1328cafaf3cdf30f2e/package/gluon-status-page) probably is the best starting point.

NeoRaider, 2015-12-10:
The old status page is still included in the gluon-status-page package ( https://github.com/freifunk-gluon/gluon/tree/master/package/gluon-status-page/files/lib/gluon/status-page/www ), just use http://[address]/cgi-bin/status to reach it.

closed 2016-08-22 by rotanid