freifunk-gluon/gluon #600: manifest file isn't cleaned with image directory

SimJoSt, 2015-12-13:
Building a different branch after already building another one, leaves the manifest file of the previous build uncleaned.
Example:
first build = stable
`images/sysupgrade` contents = stable image files and stable.manifest
second build = testing
`images/sysupgrade` contents = testing image files, testing.manifest **and** stable.manifest

NeoRaider, 2015-12-13:
This is working as expected, this way you can create multiple manifests for a single build (in Lübeck we often use the same images for the beta upgrader branch and then released them as stable later).

My recommendation would be to use a distinct output dir for each build if you don't want this (`make ... GLUON_OUTPUTDIR=foo`).

closed 2015-12-13 by NeoRaider

SimJoSt, 2015-12-14:
I understand. Having multiple manifests for one build might be nice.
But having 2 manifests, one for the current build and one left over from the previous one, seems unnecessary. If I start a new build, I don't need the old manifests anymore, don't I?
Or am I missing something? Maybe I just don't understand in which way you create multiple manifests.