freifunk-gluon/gluon #602: ATH9K Problems in ubiquiti devices

A-Kasper, 2015-12-13:
There are ATH9K Driver Problems in ubiquiti devices (Bullet M).
Here are some logs from different Devices:

[   37.750000] random: nonblocking pool is initialized
[41587.000000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[47288.840000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[50733.260000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[50733.290000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[59837.410000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[61775.880000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[63045.070000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[67689.420000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[68547.630000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[69235.290000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[69731.540000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[75229.940000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[78369.820000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[81355.440000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[82701.910000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[131497.820000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[132208.180000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[133165.330000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[134208.380000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0
[144707.290000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[146382.370000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0
[151419.980000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[153242.410000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[161760.600000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[165363.380000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[172255.850000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[184705.310000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[188027.130000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[201023.380000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[209985.330000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[210524.910000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[218874.940000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[218898.690000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[223763.550000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[226986.340000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[227833.090000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[236168.240000] ath: phy0: unsupported hw bitrate detected 0x12 using 1 Mbit
[239577.470000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[241491.780000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[242211.700000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[242571.540000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[243396.200000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[257650.600000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[260302.000000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[271650.330000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[279576.440000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[279594.150000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[303769.690000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[319893.950000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[331693.640000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[335463.550000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[337925.090000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[339051.260000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[344186.940000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0
[402202.020000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[403155.250000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0
[406332.790000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[406913.410000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[408429.370000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[409053.110000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[413854.230000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[414059.860000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[419701.720000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[448261.180000] br-wan: received packet on eth0 with own address as source address
[448261.190000] br-wan: received packet on eth0 with own address as source address
[448261.200000] br-wan: received packet on eth0 with own address as source address
[... 298 lines cut ...]
[466436.760000] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat sch_teql sch_sfq sch_red sch_prio sch_pie sch_netem sch_htb sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc batman_adv libcrc32c ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6t [... cut ...]
[583951.960000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[593763.240000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[631027.640000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[631027.680000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[755852.540000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0

A-Kasper, 2015-12-13:
this is the log from a 1043... it's another error

root@FF-BO-Sockel-Braun-SW:~# dmesg
[    0.000000] Linux version 3.18.23 (root@webserver.derherzen.de) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r47335) ) #14 Mon Dec 7 01:09:38 CET 2015
[    0.000000] MyLoader: sysp=ffffffff, boardp=ffffffff, parts=ffffffff
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x03ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803802b0, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=TL-WR1043ND-v2 console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 60812K/65536K available (2615K kernel code, 132K rwdata, 548K rodata, 216K init, 194K bss, 4724K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:720.000MHz, DDR:600.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] NET: Registered protocol family 16
[    0.080000] MIPS: machine is TP-LINK TL-WR1043ND v2
[    0.520000] Switched to clocksource MIPS
[    0.530000] NET: Registered protocol family 2
[    0.530000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.530000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.540000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.540000] TCP: reno registered
[    0.550000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.550000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.560000] NET: Registered protocol family 1
[    0.560000] PCI: CLS 0 bytes, default 32
[    0.560000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.580000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.580000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.590000] msgmni has been set to 118
[    0.600000] io scheduler noop registered
[    0.600000] io scheduler deadline registered (default)
[    0.600000] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.610000] console [ttyS0] disabled
[    0.640000] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.640000] console [ttyS0] enabled
[    0.650000] bootconsole [early0] disabled
[    0.660000] m25p80 spi0.0: found s25fl064k, expected m25p80
[    0.670000] m25p80 spi0.0: s25fl064k (8192 Kbytes)
[    0.670000] 5 tp-link partitions found on MTD device spi0.0
[    0.680000] Creating 5 MTD partitions on "spi0.0":
[    0.680000] 0x000000000000-0x000000020000 : "u-boot"
[    0.690000] 0x000000020000-0x000000142fdc : "kernel"
[... 141 lines cut ...]
[12894.270000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[13802.820000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[18463.460000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[36071.590000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[73083.380000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[102266.250000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[162050.510000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[165659.390000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[166000.280000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[166256.920000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[166447.840000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[174179.670000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[... cut ...]
[373818.500000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[376459.360000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[509545.980000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[520443.790000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[523845.130000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a
[566378.740000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xc055a

NeoRaider, 2015-12-13:
How do these messages correlate with actual issues? Do you sometimes or always see issues when these messages occur? Do you always see such messages when there are WLAN issues?

If you have actual issues, what kind are these?
- Can the WLAN still be seen?
- Does reception still work? (Does a iw station dump still see other WLANs?)
- Are both AP and IBSS/11s affected?
- Anything else you have found out about the nature of the issues

A-Kasper, 2015-12-14:
Hi Matthias,

Am 14.12.2015 um 00:51 schrieb Matthias Schiffer:
> How do these messages correlate with actual issues? Do you sometimes or
> always see issues when these messages occur? Do you always see such
> messages when there are WLAN issues?

The WLAN is unstable. You can see it, you can connect, but very often
you don't get any data.

> If you have actual issues, what kind are these?
> 
>   * Can the WLAN still be seen?

yes

>   * Does reception still work? (Does a iw station dump still see other
>     WLANs?)
The known meshnodes are in the city-center of Bochum. There are always
more than one possible node I'm connected to, so I wasn't able to
investigate the problems very deep. A look in dmesg shows some driver
messages

>   * Are both AP and IBSS/11s affected?

In this case it's an 11s only setup. I can't say if IBSS is also affected.

>   * Anything else you have found out about the nature of the issues

Needs deeper investigation. I will have a look asap.
Maybe some other testers could tell if they could find similar errors
and messages.

switch42, 2015-12-14:
Hi

I see this messages in my "Ubiquiti PicoStation M2", but they are not so often:

[160630.120000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[259506.210000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[261612.410000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[281808.490000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[293237.710000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[308144.970000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[379030.580000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000082c0
[450644.400000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[548319.750000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[549620.250000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[570665.340000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[570717.460000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[624439.820000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[640241.670000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[656776.420000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[656776.440000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[811768.410000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000042c0
[814064.470000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[823373.380000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[861493.360000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[913840.910000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0x40d22
[962630.400000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
[1025259.540000] ath: phy0: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x42100020 DMADBG_7=0x000084c0
 15:03:01 up 13 days,  7:26,  load average: 0.25, 0.20, 0.20
root@FF-BO-WAT-Hoentrop-S:~#

When I read the log correctly I hav the first problem after 160630 seconds uptime?! 
I can try to reproduce this message, but I have no idea to do this.

A-Kasper, 2015-12-14:
Thank you. But I've to add, that you are using my images.
I would like to preclude, that the failure is just in my images because
of voodoo ;)

switch42, 2015-12-15:
Yes it's correct.
Last hint: I see this errors with ath on all my managed routers (1043nd, 901nd, 841nd,wa860re).
Maybe your are right that only your builds are infected?

NeoRaider, 2016-02-28:
The Gluon master just got an updated mac80211 backport, which was reported to fix some ath9k issues.

NeoRaider, 2016-05-16:
Closing, we're tracking ath9k issues in #605.

closed 2016-05-16 by NeoRaider