freifunk-gluon/gluon #604: Problem building v2015.1.2

marvin, 2015-12-21:
Hi Guys,

On a new machine I am trying to build the target=ar71xx-generic with our Site.conf which works.

uname: "Linux build-server 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt20-1+deb8u1 (2015-12-14) x86_64 GNU/Linux"

This is a virtual machine on a hyper-V

I am using this command to build: "make GLUON_TARGET=ar71xx-generic V=s"
and I am getting this error. Any idea?

checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... unsupported
checking for exception model to use... configure: error: unable to detect exception model
Makefile:9467: recipe for target 'configure-target-libstdc++-v3' failed
make[6]: *** [configure-target-libstdc++-v3] Error 1
make[6]: Leaving directory '/home/david/gluon/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final'
Makefile:857: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/david/gluon/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final'
Makefile:76: recipe for target '/home/david/gluon/build/ar71xx-generic/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final/.built' failed
make[4]: *** [/home/david/gluon/build/ar71xx-generic/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final/.built] Error 2
make[4]: Leaving directory '/home/david/gluon/openwrt/toolchain/gcc/final'
toolchain/Makefile:86: recipe for target 'toolchain/gcc/final/compile' failed
make[3]: *** [toolchain/gcc/final/compile] Error 2
make[3]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
toolchain/Makefile:85: recipe for target '/home/david/gluon/build/ar71xx-generic/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install' failed
make[2]: *** [/home/david/gluon/build/ar71xx-generic/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install] Error 2
make[2]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
/home/david/gluon/Makefile:289: recipe for target 'prepare' failed
make[1]: *** [prepare] Error 2
make[1]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
Makefile:69: recipe for target 'all' failed
make: *** [all] Error 2

thanks

A-Kasper, 2015-12-21:
You may try with V=ss to get more information.

marvin, 2015-12-21:
Tried that. Same result.

checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... unsupported
checking for exception model to use... configure: error: unable to detect exception model
Makefile:9467: recipe for target 'configure-target-libstdc++-v3' failed
make[6]: *** [configure-target-libstdc++-v3] Error 1
make[6]: Leaving directory '/home/david/gluon/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final'
Makefile:857: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/david/gluon/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final'
Makefile:76: recipe for target '/home/david/gluon/build/ar71xx-generic/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final/.built' failed
make[4]: *** [/home/david/gluon/build/ar71xx-generic/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final/.built] Error 2
make[4]: Leaving directory '/home/david/gluon/openwrt/toolchain/gcc/final'
toolchain/Makefile:86: recipe for target 'toolchain/gcc/final/compile' failed
make[3]: *** [toolchain/gcc/final/compile] Error 2
make[3]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
toolchain/Makefile:85: recipe for target '/home/david/gluon/build/ar71xx-generic/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install' failed
make[2]: *** [/home/david/gluon/build/ar71xx-generic/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/stamp/.toolchain_install] Error 2
make[2]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
/home/david/gluon/Makefile:289: recipe for target 'prepare' failed
make[1]: *** [prepare] Error 2
make[1]: Leaving directory '/home/david/gluon/build/ar71xx-generic/openwrt'
Makefile:69: recipe for target 'all' failed
make: *** [all] Error 2

jplitza, 2015-12-21:
Does "new machine" mean that you have a freshly cloned gluon repository? If not, did you execute `make dirclean`?
Please post the complete build log, not only excerpts, and use some nopaste service (like [Gist](https://gist.github.com) for example). Also, please nopaste the file `/home/david/gluon/openwrt/build_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04-final/config.log`.

marvin, 2015-12-21:
No "new machine" means we have a different machine where we build our images.
I've build the x86-generic on it without problems not i am trying to build the ar71xx-generic.
Yes ill post the stuff in a few minutes. thanks

marvin, 2015-12-21:
@jplitza Thx "make dirclean" was the solution I think. Because i've built x86-generic before and havent fully cleaned all the directories.

thx

closed 2015-12-21 by marvin

jplitza, 2015-12-21:
Normally, different architectures should have different toolchain directories and thus they shouldn't have interfered. Maybe an old build of the same architecture failed and left broken pieces behind? Anyway, great if it's working now.