freifunk-gluon/gluon #605: WifiStack Problem in 2012.2?

AKA-47, 2015-12-22:
Das Problem wird aktuell hier im Forum diskutiert: 

https://forum.freifunk.net/t/wifi-mesh-probleme-im-2012-2-gluon/9813/14

Die Konstellation:

    Mehrere 841v10 mit 2012.2 im WIFI Mesh
    getestet mit selbst gebauter Firmware und experimental builds anderer Site (also nicht von mir)
    getestet zusätzlich mit 1043v2
    KEIN 802.11s!!
    Immer nur einen Knoten mit VPN

Das Problem:

Sobald ich 2 Router mit 2012.2 im Mesh habe bricht mir das WIFI mesh soweit ein das kaum noch Pakete durchgehen.

Folgende Test wurden gemacht um das Problem einzugrenzen:

Eigene Firmware:
-2er Mesh mit 2012.1 auf 1043 und 2012.2 auf 841v10 --> OK
-2er Mesh mit 2012.2 auf 841v10 --> Problem tritt auf
-2er Mesh mit 2012.2 auf 1043er und 2012.2 auf 841v10 --> Problem tritt auf
-2er Mesh mit 2012.2 auf 1043er --> Problem tritt auf
-3er Mesh mit 2012.1 auf 1043 und 2012.2 auf ZWEI 841v10 --> Problem tritt auf

Andere Firmware:
-2er Mesh mit 2012.1 auf 1043 und 2012.2 auf 841v10 --> OK
-2er Mesh mit 2012.2 auf 841v10 --> Problem tritt auf
-2er Mesh mit 2012.2 auf 1043er und 2012.2 auf 841v10 --> Problem tritt auf
-2er Mesh mit 2012.2 auf 1043er --> Problem tritt auf
-3er Mesh mit 2012.1 auf 1043 und 2012.2 auf ZWEI 841v10 --> Problem tritt auf

Fazit: Identische Ergebnisse der Testfälle unabhängig der Hardware! Somit kann ich also den 841v10 als Fehlerquelle sowie auch meine gebackene Firmware ausschließen.
Ich habe sogar Testweise mal die WIFI Kanäle gewechselt was aber auch nichts gebracht hat

Geflasht wurde sicherheitshalber IMMER ein Factory Image mit TFTP Recovery um sicherzugehen das die Kisten komplett neu sind.

Evtl kann sich ein Entwickler daz äußern?

NeoRaider, 2015-12-22:
1. Ist noch ein direkter Ping über das Link-local auf dem ibss0-Interface möglich? Bitte sowohl Unicast (fe80::......%ibss0) als auch Multicast (ff02::1%ibss0) testen.
2. Wie sieht `iw dev ibss0 station dump` aus? Bitrate? Signalpegel?
3. Von batman-adv gemessene TQ zu den Nachbarknoten (batctl o)?

Uns sind bisher keine solchen Probleme aufgefallen.

AKA-47, 2015-12-22:
1)
ping6 fe80::60e6:28ff:febe:18c4
PING fe80::60e6:28ff:febe:18c4 (fe80::60e6:28ff:febe:18c4): 56 data bytes
64 bytes from fe80::60e6:28ff:febe:18c4: seq=0 ttl=64 time=0.573 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=1 ttl=64 time=0.419 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=2 ttl=64 time=0.401 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=3 ttl=64 time=0.480 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=4 ttl=64 time=0.494 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=5 ttl=64 time=0.394 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=6 ttl=64 time=0.397 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=7 ttl=64 time=0.401 ms
64 bytes from fe80::60e6:28ff:febe:18c4: seq=8 ttl=64 time=0.477 ms
^C
--- fe80::60e6:28ff:febe:18c4 ping statistics ---
9 packets transmitted, 9 packets received, 0% packet loss
round-trip min/avg/max = 0.394/0.448/0.573 ms

ping6 ff02::1
PING ff02::1 (ff02::1): 56 data bytes
64 bytes from fe80::62e3:27ff:febe:18c4: seq=0 ttl=64 time=1.045 ms
64 bytes from fe80::62e3:27ff:febe:18c4: seq=1 ttl=64 time=0.664 ms
64 bytes from fe80::62e3:27ff:febe:18c4: seq=2 ttl=64 time=0.643 ms
64 bytes from fe80::62e3:27ff:febe:18c4: seq=3 ttl=64 time=0.722 ms
64 bytes from fe80::62e3:27ff:febe:18c4: seq=4 ttl=64 time=0.581 ms
--> alles super

2) 
Node2:
iw dev ibss0 station dump
Station 62:e6:28:c6:d7:ee (on ibss0)
	inactive time:	0 ms
	rx bytes:	3504727
	rx packets:	28048
	tx bytes:	70611
	tx packets:	795
	tx retries:	104
	tx failed:	0
	signal:  	-38 [-39, -44] dBm
	signal avg:	-37 [-39, -43] dBm
	tx bitrate:	130.0 MBit/s MCS 15
	rx bitrate:	117.0 MBit/s MCS 14
	expected throughput:	4.503Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no

Node1:
Station 62:e6:28:be:18:c4 (on ibss0)
	inactive time:	30 ms
	rx bytes:	232907589
	rx packets:	1909673
	tx bytes:	4257570
	tx packets:	25132
	tx retries:	8100
	tx failed:	47
	signal:  	-37 [-41, -40] dBm
	signal avg:	-37 [-40, -40] dBm
	tx bitrate:	65.0 MBit/s MCS 7
	rx bitrate:	1.0 MBit/s
	expected throughput:	3.194Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no

3)
batctl o | grep ibss
62:e6:28:be:18:c4    0.170s   (201) 62:e6:28:be:18:c4 [     ibss0]: 62:e6:28:be:18:c4 (201)

Hatte ich alles schon durchgecheckt. und konnte das Problem sogar an verschiedenen Standorten reproduzieren... bin ein wenig ratlos

AKA-47, 2015-12-22:
Achso...und die andere Seite:
Node2:
batctl o | grep ibss | grep "62:e6:28:c6:d7:ee  "
62:e6:28:c6:d7:ee    0.450s   (209) 62:e6:28:c6:d7:ee [     ibss0]: 62:e6:28:c6:d7:ee (209)

NeoRaider, 2016-02-28:
The Gluon master just got an updated mac80211 backport, which was reported to fix some ath9k issues.

AKA-47, 2016-03-02:
Sehr gut! Jetzt geht alles wie es soll! Vielen Dank

AKA-47, 2016-03-11:
Wann kommt dieser commit in die stable?

jplitza, 2016-03-11:
Mit 2016.2. Für ein 2016.1.x-Release ist diese Änderung denke ich zu groß. Und 2016.2 braucht vermutlich noch eine Weile (siehe [Milestone](https://github.com/freifunk-gluon/gluon/milestones/2016.2))

NeoRaider, 2016-03-24:
Ich bin am Überlegen, nach v2016.1.3 einen größeren Backport zu machen und in 1-2 Wochen als v2016.1.4 zu releasen, um solche Probleme zu beheben.

A-Kasper, 2016-03-25:
Am 24.03.2016 um 22:48 schrieb Matthias Schiffer:
> Ich bin am Überlegen, nach v2016.1.3 einen größeren Backport zu machen
> und in 1-2 Wochen als v2016.1.4 zu releasen, um solche Probleme zu beheben.

Ich fänd das sehr sehr gut und wichtig. Ich hatte sogar schon überlegt
auf die 2015er zurück zu gehen, weil die wireless geschichten sich so
krass verschlechtert haben. Das ist schon eine sehr wichtige Funktionalität.

NeoRaider, 2016-05-16:
Ich habe einen neuen Branch *v2016.1.x-mac80211-test* hinzugefügt, der ein weiters Update der WLAN-Treiber enthält. Das ganze besiert auf v2016.1.x, damit ein unkompliziertes Wechseln zwischen Releases und dem Test-Branch möglich ist.

Es wäre sehr hilfreich, wenn die neue Version auf möglichst vielen Routern mit WLAN-Problemen getestet wird.

NeoRaider, 2016-05-20:
*v2016.1.x-mac80211-test* hat gerade ein weiteres Update bekommen, bitte testen.

rotanid, 2016-05-21:
update heute nacht auf ein paar Knoten gespielt, schon jetzt hat einer davon wieder den Bug :-(
geringe sample size, würde aber bisher eher sagen es ist schlechter als ohne den test-branch

oszilloskop, 2016-05-23:
Das Problem tritt mit der v2016.1.x-mac80211-test weiterhin auf.
Bei mir sieht es so aus, als ob sich die v2016.1.4 und die v2016.1.x-mac80211-test bezüglich der Problemhäufigkeit nichts geben.

NeoRaider, 2016-06-07:
v2016.1.x-mac80211-test enthält jetzt einen neuen Patch, bitte testen.

NeoRaider, 2016-06-07:
Ähm... der neue Patch kompiliert noch nicht, einen Moment...

NeoRaider, 2016-06-07:
So, v2016.1.x-mac80211-test enthält jetzt 2 Commits über v2016.1.x hinaus: Ein weiteres Update von mac80211 und ein Patch von nbd.

Bitte erstmal nur das Update testen (also den zweitneusten Commit). Wenn das keinen Unterschied macht, dann einmal mit dem neusten.

NeoRaider, 2016-06-12:
Gibt's irgendwas neues hier?

oszilloskop, 2016-06-12:
Ja, bin gerade mit meinen Test bezüglich d31c1c9 durch.
Leider tritt das Problem weiterhin auf.

Werde heute noch 548cf1d auf einige Router aufspielen.

oszilloskop, 2016-06-14:
Getestet habe ich auf meshenden CPE210 v1.0, WR841 v8, v9 und v10 mit starker Datenlast auf dem Mesh- und dem Clientnetz.

Ich habe per Cronjobs und dem Befehl  `/sbin/wifi` ein eigenes kleines Problem-Abfangszenario inkl. Loging gebastelt. Dem trau ich aber nicht so ganz.

Nach zwei Tagen 548cf1d Test sieht es erstmal deutlich besser aus als mit d31c1c9. Statt täglichen Ausfallerscheinungen, hatte ich bisher insgesamt nur zwei Ausfälle. Ob die Zwei Ausfälle problembedingt sind, weiss ich noch nicht genau. Manchmal sind da schnelle Finger an den Ein-/Ausschaltern der Router ,und das beeinflusst mein Abfangszenario.)

Werde weiter beobachten...

oszilloskop, 2016-06-14:
Hm, wie es aussieht, so hat sich nur die Häufigkeit verringert.
Ich habe jetzt noch 2 Knoten, welche ein/das Problem haben.

Nennen wir einen Problem-Knoten mal **KPUTT**
KPUTT hat die MAC `ea:97:f7:a1:bf:2c`

**Chekov** ist ein Knoten, der meshend den Knoten KPUTT in die Wolke einbindet.

Der Befehl `iw dev ibss0 station dump` auf Chekov ergibt die unten aufgeführte Ausgabe.

Auffällig ist:
- Der Name von KPUTT kann nicht aufgelöst werden
- Die TX Bitrate von KPUTT hängt auffällig bei 1.0 MBit/s
- KPUTT (ea:97:f7:a1:bf:2c) taucht 5 mal auf)
- Auf unserer MAP liegt die TX TQ von KPUTT zu Cjekov bei 2-4%
- Auf unserer MAP ist die TX TQ von Chekov zu KPUTT nicht mehr aufgeführt

Ausgabe: 

```
root@Chekov:~# iw dev ibss0 station dump
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	6800 ms
	rx bytes:	782524109
	rx packets:	5289540
	tx bytes:	0
	tx packets:	0
	tx retries:	362968
	tx failed:	7023
	signal:  	-53 [-55, -57] dBm
	signal avg:	-53 [-55, -58] dBm
	tx bitrate:	1.0 MBit/s
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	43.303Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	23890 seconds
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	6800 ms
	rx bytes:	782486698
	rx packets:	5289265
	tx bytes:	0
	tx packets:	0
	tx retries:	362968
	tx failed:	7023
	signal:  	-53 [-55, -57] dBm
	signal avg:	-53 [-55, -58] dBm
	tx bitrate:	1.0 MBit/s
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	4.119Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	no
	MFP:		no
	TDLS peer:	no
	connected time:	23889 seconds
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	6800 ms
	rx bytes:	782487085
	rx packets:	5289266
	tx bytes:	0
	tx packets:	0
	tx retries:	362968
	tx failed:	7023
	signal:  	-53 [-55, -57] dBm
	signal avg:	-53 [-55, -58] dBm
	tx bitrate:	1.0 MBit/s
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	4.119Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	no
	MFP:		no
	TDLS peer:	no
	connected time:	23889 seconds
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	6800 ms
	rx bytes:	782487085
	rx packets:	5289266
	tx bytes:	0
	tx packets:	0
	tx retries:	362968
	tx failed:	7023
	signal:  	-53 [-55, -57] dBm
	signal avg:	-53 [-55, -58] dBm
	tx bitrate:	1.0 MBit/s
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	4.119Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	no
	MFP:		no
	TDLS peer:	no
	connected time:	23889 seconds
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	30 ms
	rx bytes:	804896639
	rx packets:	5668484
	tx bytes:	2285864334
	tx packets:	2674955
	tx retries:	362968
	tx failed:	7023
	signal:  	-54 [-56, -58] dBm
	signal avg:	-53 [-55, -57] dBm
	tx bitrate:	117.0 MBit/s MCS 14
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	43.303Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	23889 seconds
```

oszilloskop, 2016-06-14:
Nochmal ich.
Nachdem ich auf **Chekov**! den Befehl `wifi` abgesetzt habe, sieht die Ausgabe von 
`root@Chekov:~# iw dev ibss0 station dump` wie folgt aus:
```
Station ea:97:f7:a1:bf:2c (on ibss0)
	inactive time:	40 ms
	rx bytes:	157287
	rx packets:	2259
	tx bytes:	345672
	tx packets:	4342
	tx retries:	1050
	tx failed:	54
	signal:  	-54 [-55, -58] dBm
	signal avg:	-53 [-55, -58] dBm
	tx bitrate:	104.0 MBit/s MCS 13
	rx bitrate:	104.0 MBit/s MCS 13
	expected throughput:	40.283Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	143 seconds 
````

`batctl o | grep ea:97:f7:a1:bf:2c` ergibt folgendes:
ea:97:f7:a1:bf:2c    2.330s   (  7) ea:97:f7:a1:bf:2c [     ibss0]: ea:97:f7:a1:bf:2c (  7)

**KPUTT ist weiterhin nicht zu erreichen.**

Hm, ist das jetzt das Problem von einem Router, oder des Zusammenspiel von beiden Routern?

NeoRaider, 2016-06-14:
Ok, eine vollständige Lösung aller Probleme wäre auch zu schön gewesen. 2 weitere Tests wären hilfreich:

1. In 548cf1d gibt es die Datei `openwrt/package/kernel/mac80211/patches/990-test.patch`; diese enthält 2 unabhängige Patch-Chunks. Diese beiden Chunks sollten einmal unabhängig voneinander getestet werden, um herauszufinden, welcher sich auf das Problem auswirkt. Nach Verändern der Patch-Datei einfach einmal `make package/mac80211/clean` ausführen, damit es neu gebaut wird.

2. Ich hatte die Ursache des Problems bisher auf eine Version zwischen compat-wireless 2015-03-09 und 2015-07-21 eingegrenzt, da nach dem Update auf 2015-07-21 vermehrt Bug Reports kamen. Was sind deine beiden Problemknoten für Hardware? Wenn die Hardware schon von 6e79982442b17de54a52ad4fc5e7ba1558475412 unterstützt wurde, wäre auch damit ein Vergleich hilfreich, um herauszufinden, ob das Problem auch unter 2015-03-09 auftritt.

oszilloskop, 2016-06-14:
Die Problemknoten sind WR841 v8 und v10.

Okay, werde versuchen das alles mal irgendwie durchzutesten. Das wird mehrere Tage dauern.

oszilloskop, 2016-06-20:
Aufgrund, dass ich nur eingeschränkten physischen Zugriff auf einige Problem Router habe, habe ich
vor mehrerenTagen nur den Commit 6e79982 aus Punkt 2. auf 12 Unterkunfts-Router aufgespielt (WR841 v8,V9 und V10). 

- Das Fehlerbild  der reduzierten Mesh TX-TQ ist innerhalb von 5 Tagen noch 3 mal aufgetreten (leider an einem wichtigen Mesh Knoten). 
- Das Fehlerbild, das sich das WLAN komplett verabschiedet hat, trat bei mir noch 1 mal auf.

Ob und wann ich die Untersuchung aus Punkt 1. durchführen kann, ist noch nicht wirklich klar.

oszilloskop, 2016-06-21:
Gerade hatte ich bei einem WR841 v8 mit Gluon **v2016.1.5** einen Mesh-Hänger. Die meshende Gegenstelle sah den Router mit einer TQ von nur 5%.<br>
Nachdem ich die Gegenstelle mit dem Befehl 'wifi' bearbeitet hatte, sah die Gegenstelle den Problem-Router (er tauchte auf der Statusseite auf), jedoch wurde als TQ nur ein '-' angezeigt.

Da der hängende Router per cronjob alle paar Stunden 'wifi' aufruft, konnte ich nach der Wiederbelebung des WLANs folgendes per 'dmesg' auslesen:<br>
Hier klicken ->  http://pastebin.com/0Re0uXJ0

Ab Zeil 231 wird es evtl. interessant.

NeoRaider, 2016-06-22:
@oszilloskop, wie lange wartest du nach so einem Reset, bis du die TQ anschaust?

Allgemein dauert es etwa 10 Minuten, bis die TQ von einem neuen Link ihren endgültigen Wert erreicht hat, da die TQ aus einem gleitenden Mittelwert des Paketverlusts gebindet wird. Dazu kommt, dass eine TQ nur dann überhaupt von batman-adv ausgespuckt wird, wenn die direkte Verbindung bevorzugt ist. Wenn es also einen Zwischen-Hop gibt und batman-adv die Verbindung über den zusätzlichen Hop sinnvoller findet, wird als TQ für die direkte Verbindung immer nur '-' angezeigt.

Die Kernel-Meldung ist seltsam, ich würde am ehesten auf doppelte MAC-Adressen irgendwo in dem Setup tippen, aber ich bin nicht sicher...

Du hast geschrieben, dass es mit 6e79982 noch ein paar Mal Probleme gab. Kannst du irgendwie in Relation setzen, wie häufig die Probleme mit den verschiedenen Versionen sind? Tritt es mit 548cf1d etwa so häufig auf wie mit 6e79982, und mit v2016.1.x deutlich häufiger?

oszilloskop, 2016-06-23:
@NeoRaider 
Das mit der TQ ist extrems merkwürdig. Habe gerade wieder einen aktuellen Ausfall der gleichen Routers (WR851N v8). Nach einem "wifi" auf der Gegenstelle, wird auf der Gegenstelle die TQ des Problem-Router angezeigt und zwar wie folgt: Sie geht ziemlich schnell auf 17% hoch und läuft dann innerhalb von 2 Minuten auf 2% Runter. Per `batctl ping -c 10 xx:yy:zz` war der Problemreouter zu keiner Zeit anpingbar. 

Problemhäufigkeit.
v2016.1.x und 548cf1d geben sich nicht viel. Bei meinen Test sah es erst gut aus, nach 2-3 Tagen ist die Häufigkeit dann aber gleichgezogen und trat täglich bei mehreren Routern auf.

Der ältere 6e79982 sieht deutlich besser aus, ist aber auch nicht total clean. Häufigkeit deutlich unter 1  pro Tag und das auch auf weniger Routern.

Hm. Ich hätte Protokoll führen sollen :o( und wünschte mir, es gibt noch weitere Tester :o)

rotanid, 2016-06-23:
as we have our workaround script in place i can't help much.
today i experienced another case that our script didn't cover so far:
in "iw station dump" there was one client, which didn't appear in "batctl tl" or anywhere else. in our scripts view: "nice, there's a client, everything must be ok" - but it wasn't, wifi was not usable.
```
root@knotenname:~# iw dev client0 station dump
Station 84:1b:5e:XX:XX:XX (on client0)
	inactive time:	114290 ms
	rx bytes:	358547302
	rx packets:	1856844
	tx bytes:	33282786
	tx packets:	326875
	tx retries:	155949
	tx failed:	1533
	signal:  	-66 [-74, -67, -77] dBm
	signal avg:	-59 [-65, -64, -64] dBm
	tx bitrate:	72.2 MBit/s MCS 7 short GI
	rx bitrate:	65.0 MBit/s MCS 7
	expected throughput:	28.838Mbps
	authorized:	yes
	authenticated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	175704 seconds
```

mbaumga, 2016-06-28:
Guten Abend miteinander. Ich habe gerade einen Knoten, dessen Nachbarknoten ein kaputtes Wifi hat und keine Mesh-Verbindung zusammenbekommt. 

Der Knoten mit dem funktionierenden WLAN hat keine anderen Mesh-Partner in der Nähe, als den Knoten mit dem broken WLAN und auch keine Clients verbunden. Trotzdem erscheint bei iwinfo auf dem Mesh-Interface als Bit Rate 1.0Mbit/s. Normalerweise sollte ja bei fehlenden Verbindungspartnern die Rate auf "Unknown" wechseln.
Ich habe auf der Statusseite den kaputten Knoten mit konstanten 4% TQ als Mesh-Partner auf dem ibss0-Interface.

`root@knotenname:~# iwinfo ibss0 info
ibss0     ESSID: "mesh.tecff"
          Access Point: 02:0F:8F:XX:XX:XX
          Mode: Ad-Hoc  Channel: 6 (2.437 GHz)
          Tx-Power: 14 dBm  Link Quality: 49/70
          Signal: -61 dBm  Noise: -95 dBm
          Bit Rate: 1.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
`

Normalerweise wird doch das Interface vor Verbindungsaufbau auf 1.0Mbit/s gesetzt und auch die Managementpakete damit verschickt, da ja noch kein Aushandeln der Übertragungsrate stattgefunden hat. 

https://upimg.de/img/notbrokenwifi_5pzfuh.png <- Dump, was so alles an Paketen rumfliegt auf den Mesh-Interfaces.

Summa Summarum: Bleibt der kaputte Router irgendwo zwischen Initialisierung des Verbindungsaufbaus und aushandeln der Verbindung einfach hängen?

Adorfer, 2016-06-28:
>  Ich habe auf der Statusseite den kaputten Knoten mit konstanten 4% TQ als Mesh-Partner auf dem ibss0-Interface. 

Damit beschreibst Du ziemlich genau eines der 3-4 Szenarien: Clients verschwunden, alle Meshlinks von binnen Minutenfrist von von >80% auf 2-4% runter, dort aber "stabil"... unbenutzbar.

rotanid, 2016-06-29:
hat es zwischenzeitlich jemand geschafft, das Problem gewollt zu reproduzieren durch irgendeine Form von künstlich erzeugtem Traffic auf einem bestimmten Übertragungsweg (ibss, client, ...)?
Ich leider nicht...

bitboy0, 2016-07-11:
Ich weiß nicht ob ich das gleiche Problem hier haben könnte...
ffrn.de / gluon-v2016.1.5 stable und die versionen davor auch (seit 3 Monaten)

Mein Uplink-Knoten (1043v2) ist die ganze Zeit über online, aber alles dahinter ist tot. Auch kann man sich nicht einwählen. Ich hab jetzt einen Offloader hingestellt und da dran zwei Knoten. Einen 841V10, und den 1043v2.

Heute sah das dann so aus: http://s.ffrn.de/dashboard/db/nodes?var-Node=Ladenburg-OHS-001&var-Node=Ladenburg-OHS-002&var-Node=Ladenburg-OHS-003&var-Node=Ladenburg-OHS-004&var-Node=Ladenburg-OHS-005&var-Node=Ladenburg-OHS-006&var-Node=Ladenburg-OHS-T01&from=1468214317777&to=1468229889353

Beide Knoten (OHS-001 und OHS-T01) haben ca. 10 m zu OHS-002 und wenn nur einer der beiden ausfällt, müsste alles weiterlaufen. Aber BEIDE haben einfach für ne knappe Stunde nichts per WLAN gemacht. Keine Clients, kein Mesh. Dann ging alles wieder wie es soll.

Diese Unterbrechungen habe ich alle paar Tage und für eine Zeit von ca. 1h bis 8h.

Ab und an hatte ich auch einen Knoten der mit 5% TQ hing. Wenn ich alle Knoten außenrum neu gestartet habe, ging der wieder in's Mesh und lief dann auch wieder normal.

Sagt mir ob sich das für euch anfühlt, wie das Problem um das es hier geht?

oszilloskop, 2016-07-14:
Ja, das hört sich genau nach diesem Problem an.

NeoRaider, 2016-07-27:
Es gibt mal wieder ein mac80211-Update im Master, das einige Verbesserungen der ath9k-Stabilität bringen könnte.

Sprinterfreak, 2016-07-27:
Tritt in Regensburg auch absolut random auf.
Ich hab hier aktuell 3 Knoten (2x 1043v2, 1x CPE210) bei denen es intensiv auftritt.
In der Tat ist es durch ausführen von `wifi`zu beheben.
Drum habe ich auf betroffenen Knoten einen micron eingerichtet:

> `*/5 * * * * batctl gwl | grep '=>' >/dev/null || wifi`

Der läuft alle 5 Minuten und holt sie wieder zurück, wenn kein Gateway verfügbar/selected ist.
So bleiben sie wenigstens administrierbar...
Evtl. sollte man den dirty fix bis zum finalen fix in den Snapshot mit übernehmen?
Häufigkeit ist von 2 Minuten bis 1x alle 2 Tage alles drin... Es ist auch nicht immer wirklich von der lokalen Netzlast abhängig. Interessant wäre hier gluon-airtime.
Läuft evtl. der Hardware-Puffer vom ath bei busy Air über? -> unhandled exception?
Wie reagiert batman auf halbe/korrupte Datagramme z.B. durch busy Air?

Die Knoten meshen lediglich über wlan nicht mehr. Per lan bleiben sie erreichbar.

An einem unserer zentralen Mesh's haben wir schon in früheren Zeiten einen pingenden Pi an einem 1043 zum "Neustarten" per Relais dran. :-)
Seit dem micron ist allerdings kein Powercycel mehr nötig gewesen.

Sprinterfreak, 2016-08-01:
Der Bug ist wohl tatsächlich eher in dem ath9k-Treiber von openwrt zu suchen.
In #openwrt gibts exakt vergleichbare Reports auch ohne Gluon.

rotanid, 2016-08-01:
@Sprinterfreak übrigens, dein Cronjob ist so nicht wirklich empfehlenswert, weil er im Zweifelsfall alle 5 Minuten das wifi wieder neustartet, obwohl schon der letzte Neustart nichts gebracht hat. guck dir mal das workaround package an: https://github.com/tecff/gluon-packages/tree/master/tecff-ath9k-broken-wifi-workaround

Sprinterfreak, 2016-08-01:
@rotanid nicht schlimm. Wenn sowieso keine Gw-Verbindung besteht, darf neugestartet werden.
Ist ja auch nur ein Notfall-Einzeiler, der allerdings bisher sehr gute Ergebnisse an empfindlichen Stellen leistet. Behebt ja auch nicht das Problem, sondern nur die Auswirkungen und ist wie der broken-wifi-workaround keines Wegs eine "Lösung".

kpanic23, 2016-08-30:
So wie es aussieht, hab ich den Bug mal in flagranti erwischt, und zwar auf einem Knoten mit VPN-Uplink. So konnte ich die Logs mal auslesen, vor und nach dem Ausführen von wifi.
Ich hoffe, es hilft weiter...

[logread-nach-wifi.txt](https://github.com/freifunk-gluon/gluon/files/444765/logread-nach-wifi.txt)
[logread-vor-wifi.txt](https://github.com/freifunk-gluon/gluon/files/444766/logread-vor-wifi.txt)

NeoRaider, 2016-09-10:
Wie gut läuft ath9k zur Zeit im Gluon-Master? Besser/schlechter/ähnlich, relativ zu 2016.1.x?

rotanid, 2016-09-10:
aus dem IRC:
> 03:30:57 < hexa- > hier ist alles prima mit gluon master
> 03:31:07 < hexa- > hab keinen mesh ausfall mehr feststellen können

kpanic23, 2016-09-12:
Kann ich so unterschreiben!
Wir haben massive Probleme mit ein paar wenigen Knoten gehabt. Seit ich diese auf einen Master-Build (gluon-v2016.1-203-g91881f4) geflasht habe, ist Ruhe im Karton. Kein einziger Ausfall seither.

AKA-47, 2016-10-04:
in der 2016.2.x commit a54e7654cda80c8b9ea3fc991719c1e254c4f995 konnte ich es wieder beobachten vgl.
https://bugs.lede-project.org/index.php?do=details&task_id=176

allerdings kommt das wifi irgendwann einfach zurück. 

cat /sys/kernel/debug/ieee80211/phy0/ath9k/reset 
    Baseband Hang:  0
Baseband Watchdog: 26
   Fatal HW Error:  0
      TX HW error:  0
 Transmit timeout:  0
     TX Path Hang: 140
      PLL RX Hang:  0
         MAC Hang:  0
     Stuck Beacon: 213
        MCI Reset:  0
Calibration error:  0
Tx DMA stop error: 209
Rx DMA stop error:  0

Antaiir, 2016-10-04:
Hallo zusammen,

ich habe ebenfalls das Problem, daß einer der von mir betreuten Knoten ständig sein WLAN verliert. Es handelt sich um eine "Ubiquiti Bullet M2HP Titanium", auf welcher Gluon 2016.2 läuft.

Ich hänge mal 2 Ausgaben von logread an. Wenn ich zur Analyse noch mehr beisteuern kann, laßt es mich wissen. :)

[logread_alt.txt](https://github.com/freifunk-gluon/gluon/files/509486/logread_alt.txt)
[logread_neu.txt](https://github.com/freifunk-gluon/gluon/files/509487/logread_neu.txt)

rotanid, 2016-10-04:
hi @Antaiir , wenn du wirklich Gluon in genau Version 2016.2 laufen hast, dann wäre der erste Rat, bitte eine aktuellere Version aufzuspielen, da im Branch v2016.2.x am Freitag ein Bugfix zu diesem Problem eingegangen ist, den dein Vorposter @AKA-47 bereits benutzt.

@AKA-47 was meinst du mit "es kommt zurück" und "irgendwann"? Ist da eine gleichbleibende Zeitspanne und etwas dazu passendes im Log erkennbar beispielsweise? vermutlich bleibt da aber vorerst das LEDE-Ticket dann der geeignetere Platz für dieses Problem...

plumpudding, 2016-10-06:
Bei uns sind die CPE210 v1.0 mit gluon-v2016.2-2-ga54e765 auf einmal taub. Die Gegenseite hört den Knoten aber noch.

Adorfer, 2016-10-06:
Das Zweite (batman-mesh läuft, ipv6 nicht mehr über air erreichbar) ist eventuel kein Radio-Bug, sondern eher im Batman zu suchen. (Hier läuft dagegen der Dir bekannte Ping-check auf eine Alias-IP, die redundant auf diversen Hosts erreichbar ist, nach erstmaliger Erreichbarkeit erfolgt bei fortgesetzter Nichterreichbarkeit ein Reboot.)

plumpudding, 2016-10-08:
https://petabyteboy.de/daten/logread
https://petabyteboy.de/daten/dmesg

Antaiir, 2016-10-08:
Hallo zusammen,

ein Kollege hat für mich eine neue Firmware basierend auf dem von @rotanid erwähnten Patch gebaut und wir haben die Sache ein paar Tage beobachtet. Bis jetzt sieht alles gut aus, WLAN ist bei der besagten Bullet bisher nicht wieder weggeflogen und im Log stehen quasi nur noch diverse Handshakes drin, also ohne besondere Auffälligkeiten.

Danke. :)

VG
Antaiir

plumpudding, 2016-10-08:
Es scheint ein altes und weit verbreitetes Problem zu sein..:
https://dev.openwrt.org/ticket/7790
https://discussions.apple.com/thread/3838990?start=0&tstart=0
http://community.ubnt.com/t5/UniFi-Wireless/AP-stalls-for-10-s-log-shows-TIMEOUT-while-waiting-for-nf-to/td-p/1140374
http://community.ubnt.com/t5/UniFi-Wireless/TIMEOUT-while-waiting-for-nf-to-load/td-p/1030175
Aber ich bin hier nicht der Experte. Kann ja auch sein dass die Meldung "timeout while waiting for nf to load" nichts mit unserem Problem zu tun hat..

rotanid, 2016-10-09:
@plumpudding dann hätte man **während** der Taubheit, also vor einem reboot oder wifi befehl, den Inhalt von /sys/kernel/debug/ieee80211/phy0/ath9k/reset  noch gebraucht am Besten.
und vielleicht die Ausgabe des station dump Befehls, falls da noch Geräte auftauchen sollten trotz "Taubheit".

plumpudding, 2016-10-09:
```root@FF_GL_DRK_Jakobstr_101_UP:~# cat /sys/kernel/debug/ieee80211/phy0/ath9k/reset 
    Baseband Hang:  0
Baseband Watchdog:  0
   Fatal HW Error:  0
      TX HW error:  0
 Transmit timeout:  0
     TX Path Hang:  0
      PLL RX Hang:  0
         MAC Hang:  0
     Stuck Beacon:  0
        MCI Reset:  0
Calibration error:  0
Tx DMA stop error:  0
Rx DMA stop error:  0
root@FF_GL_DRK_Jakobstr_101_UP:~# iw dev mesh0 station dump
root@FF_GL_DRK_Jakobstr_101_UP:~# iw dev mesh0 scan
BSS f6:f7:6e:30:3c:48(on mesh0)
	TSF: 4358144384 usec (0d, 01:12:38)
	freq: 2472
	beacon interval: 1000 TUs
	capability: (0x0000)
	signal: -53.00 dBm
	last seen: 30 ms ago
	Information elements from Probe Response frame:
	SSID: 
	HT capabilities:
		Capabilities: 0x11ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-15
	HT operation:
		 * primary channel: 13
		 * secondary channel offset: no secondary
		 * STA channel width: 20 MHz
root@FF_GL_DRK_Jakobstr_101_UP:~# iw dev mesh0 station dump
Station f6:f7:6e:30:3c:48 (on mesh0)
	inactive time:	110 ms
	rx bytes:	107053
	rx packets:	813
	tx bytes:	2333
	tx packets:	17
	tx retries:	8
	tx failed:	0
	signal:  	-50 [-50, -59] dBm
	signal avg:	-49 [-50, -58] dBm
	tx bitrate:	117.0 MBit/s MCS 14
	rx bitrate:	144.4 MBit/s MCS 15 short GI
	expected throughput:	43.303Mbps
	mesh llid:	32047
	mesh plid:	56811
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	9 seconds
Station fa:32:0c:f4:fc:09 (on mesh0)
	inactive time:	110 ms
	rx bytes:	91094
	rx packets:	677
	tx bytes:	429
	tx packets:	6
	tx retries:	0
	tx failed:	0
	signal:  	-56 [-56, -68] dBm
	signal avg:	-55 [-55, -66] dBm
	Toffset:	-62128075304 us
	tx bitrate:	13.0 MBit/s MCS 1
	rx bitrate:	6.5 MBit/s MCS 0
	expected throughput:	9.612Mbps
	mesh llid:	56551
	mesh plid:	3384
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	connected time:	9 seconds
root@FF_GL_DRK_Jakobstr_101_UP:~# ```

NeoRaider, 2016-10-10:
Ich werde zusehen, ath9k wieder in den Zustand von Commit 91881f4 zu bekommen, da wahrscheinlich in der Zwischenzeit ein neues Problem aufgetreten ist.

NeoRaider, 2016-10-11:
Ich habe ein mac80211-Downgrade im Gluon-Master committed (487922a8d9b7ac564effe25632fd90e2cb6979b9). Bitte testen, wenn es läuft, kommt es nach v2016.2.x.

kpanic23, 2016-10-14:
Also ich habe eine Firmware mit diesem Stand auf ein paar Knoten ausgerollt. Einer der Mesh-Knoten hat sich gestern verabschiedet, war heute aber merkwürdigerweise wieder online, muss sich wohl irgendwie von selbst neu gestartet haben. Uptime jetzt 18 Stunden, muss also gestern gegen 23.00 Uhr passiert sein.
Heute hat sich ein anderer Knoten weggehängt, ist aber vorhin wundersamerweise wieder aufgetaucht.
Ich habe mich gleich draufverbunden und das Log gezogen. Merkwürdigerweise steht im betreffenden Zeitraum absolut gar nichts im logread-Log. dmesg zeigt:
[18422.850000] ath: phy0: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xd0dda

Logread: http://pastebin.com/pNyuKxzc

[image]

Antaiir, 2016-10-15:
Wir haben einige Knoten mit der aktuellen Version versehen. Bei einer "PicoStation M2HP" (nur Mesh) taucht seit kurzem das Problem auf, daß sie sich dauerhaft aus dem WLAN ausklinkt, sobald man an der Konsole "wifi" abgesetzt hat. Die Pico ist dann nur noch mittels Reset zu reanimieren. Auf anderen Knoten ist mir bisher noch nichts in dieser Richtung aufgefallen.

VG
Antaiir

plumpudding, 2016-11-06:
Bei uns ist seit dem Update von https://github.com/freifunk-gluon/gluon/commit/a54e7654cda80c8b9ea3fc991719c1e254c4f995 auf https://github.com/freifunk-gluon/gluon/commit/1c3d97889adbdb6b7b316174645d6c5e92ad228d alles zurück bei alter Stabilität. In den letzten 2 Tagen hat kein Router durchgebootet und es wird mindestens das doppelte an Traffic geschoben.

CodeFetch, 2016-11-07:
Könnte das vielleicht ein Kalibrierungsfehler sein? Im master von Linux sind einige Patches, die die Peak-Detect-Kalibrierung der betroffenen Chips auf „softwaretechnisch gelöst” umstellen (der hier und ganz viele andere):
https://github.com/torvalds/linux/commit/7da1ddddd55fdf478f712643e145e5d343f4ba46#diff-722db64892e310f0729be9025924ba52

Ich teste schon seit ein paar Tagen alle möglichen Patches, die noch nicht bei LEDE oder OpenWrt eingepflegt sind, aber ich dachte NeoRaider wäre da schon dran... Übermorgen werde ich mal berichten, wenn sich hier nichts anderes ergibt.

Ähnliche Fehler gab es nämlich bei dem WR940N v3, aber ich bin mir nicht sicher, ob die Ursache die gleiche ist. Es könnte ja sein, dass mit der Einführung der zusätzlichen Energiesparfunktionen auf einmal die Hardwarekalibrierung weiterer Chips nicht mehr funktioniert (diese Funktionen schaffen übrigens Airtime und sie abzuschalten ist nicht im Freifunksinne und es hat bei den betroffenen Communities das Problem auch nicht gelöst).

Wieso ist hier überhaupt alles auf Deutsch?

viisauksena, 2016-11-13:
as mentioned in the other Ticket ( #993 and maybe #889 ) 
at least with release v2016.2.x there is a light tendency for ibss0 (or whole wifi?) to fail/hickup .. being unresponsive
this is also mentionend in https://forum.freifunk.net/t/announce-gluon-v2016-2-1 and https://forum.freifunk.net/t/wifihaenger-in-gluon-2016-2-x-quickfix/13821
a working solution seem to be `wifi` (think @rotanid mentioned this ) or done successfull by myself `iwinfo phy0 scan`

here some lines from closed other tickets.. sorry for mixing denglish together

@rotanid du wolltest konkretes feedback,
**Firmware** mit dem master build zu dem zeitpunkt wo der 2016.2.1 rauskam, inkl make clean und make update im vorfeld. benutzt wird ibss0 und bat-v14 - aufgefallen an 2 unterschiedlichen Tplink 841 - die autoupdate hinter sich haben.

**symptom** : wegbrechen der mesh-links - simply dead ... ifconfig mit normaler ausgabe, nichts im logread, nichts im dmesg, aufgefallen weil der mit uplink (!) in einem dichten Meshnetz ohne Mesh-Kontakte war.

**Lösung** - live beobachtet über die status page (die war bei ibss0 leer) - "iwinfo phy0 scan" .. danach sofort neu gefundene meshpartner in der status page. und logisch im meshviewer nach der entsprechenden Verzögerung.
gelöst wurde das temporär mit einem Stündlichen iwinfo phy0 scan als micron job
daher der Bugname "hickup".

ich weis das @Adorfer schon früher von sowas berichtet hatte, konnte das aber nie so genau beobachten. bei uns (Freiburg-support branch) kann das aber aufgrund von sicherungsscripten NUR bei uplinkroutern passieren, alle anderen verlieren ja durch wegbrechendes Mesh ihren uplink und würden final sogar rebooten (nach fastd, network, restart versuchen) >> https://github.com/viisauksena/gluon-fffr/blob/master/files/lib/gluon/fffr/emergency.sh (das löst nicht das mesh link problem, aber durch restart/reboot eben indirekt schon)

viisauksena, 2016-11-13:
weis nicht ob das zusammenhängt, aber eine "seltsame" beobachtung kurz hier notiert:
an einem meshenden laufenden router ohne symptome
`iwinfo phy0 scan ;dmesg |tail -n 2; logread | tail -n2`
erzeugt VOR dem output von iwinfo noch 
```
[11062.610000] IPv6: ADDRCONF(NETDEV_UP): tmp.phy0: link is not ready
Sun Nov 13 19:45:50 2016 kern.info kernel: [10964.770000] IPv6: ADDRCONF(NETDEV_UP): tmp.phy0: link is not ready
```
aber das vielleicht einfach nur ein relikt weil phy0 kein direktes if ist, tauscht man das mit ibss0 client0 oder so - gibts keinen fehler

[16 more replies]

closed 2017-05-09 by NeoRaider