freifunk-gluon/gluon #606: Android Browser and the new status page

JamesT42, 2015-12-27:
Hi!
I'm testing the current master on an offloader and when viewing the page with the 2 Android devices I own, i don't see all information.
However, the page works finde on the same device with Opera. So I do not think it is an ipv6 connection issue.
View on PC: https://www.dropbox.com/s/tx6d9yrehek8oj5/PC-Ok.jpg?dl=0
View on Smartphone with Android 4.4.4: https://www.dropbox.com/s/a8m2twmtvmjfpup/Phone-Android-4.4.4.jpg?dl=0 Missing some mesh infos
View on Tablet with Android 4.4.2: https://www.dropbox.com/s/a0u46gsqg3fgaci/Tablet-Android-4.4.2.png?dl=0 Missing all mesh infos
(It is an offloader, so the lack of signal strength for the mesh devices is due to mesh over lan)
Cheers,
JT

tcatm, 2015-12-27:
Currently only (recent versions of) Firefox and Chrome are supported. Other browser might experience issues. In most cases it doesn't work due to these not supporting a feature called "Server-sent events".

Byggvir, 2016-02-06:
Currently my Firefox on Android doesn't show Systemlast, RAM, NVRAM and Traffic. Chrome shows these value and they are updated. But Firefox is instable and unusable at the moment. I would not bother with Firefox at the moment.

JamesT42, 2016-02-06:
I have not seen any issues with Firefox.

jplitza, 2016-02-07:
I can confirm @Byggvir's report. My Firefox for Android doesn't show these values either. I does show the graphs and everything else, though.

@JamesT42 You did see that this is about Firefox on *Android*, right?

JamesT42, 2016-02-07:
@jplitza Yeah sorry, didnt read the Android part. In Firefox Android I can confirm the problem.

tcatm, 2016-02-08:
Firefox for Android is missing [Internalization API](http://caniuse.com/#feat=internationalization).

commit 8c4c6fe0 (2016-02-24): gluon-status-page: don't use internationalization API

It is not supported by all browsers yet.

Also change load display always to display 2 decimal digits (as it is read
from the kernel with 2 decimal digits).

Fixes #606
commit 6e452f20 (2016-02-29): gluon-status-page: don't use internationalization API

It is not supported by all browsers yet.

Also change load display always to display 2 decimal digits (as it is read
from the kernel with 2 decimal digits).

Fixes #606
commit 35922740 (2016-02-29): gluon-status-page: don't use internationalization API

It is not supported by all browsers yet.

Also change load display always to display 2 decimal digits (as it is read
from the kernel with 2 decimal digits).

Fixes #606
closed 2016-02-29 by NeoRaider