freifunk-gluon/gluon #608: ubiquiti-airgateway: wrong interface assignment

corny, 2015-12-30:
We tested the current master on a ubiquiti-airgateway. Currently the LAN port (eth0) is used for br-client, the POE port (eth1) is used for br-wan. The correct assignment should be instead:
* br-wan: eth0
* br-client: eth1

NeoRaider, 2016-01-04:
Sounds normal to me. What are the labels written next to the ports?

We usually assign the PoE port to WAN. Making this configurable is on our TODO list (#284).

Adorfer, 2016-01-06:
@NeoRaider  please have look at 
https://www.wifianten.com/images/product_detail_images/airgateway_poe.jpg

This target "airgatewy" is plugged usually between the poeinjector and the outdoor(nano-)station. 
So the POE interface is pointing towards the other Gluon device. 
So you usually want to have the mesh interface towards the other gluon device (even though none of the devices should under normal circumstances run fastd due to lack of cpu power, only L2TP could be an option)

kaechele, 2016-01-06:
Indeed, plugging the airGateway PoE output into hardware that doesn't like Passive PoE on it's ports may result in fried hardware :fireworks: 
But then again the device on the other side (probably airMAX or UniFi) may also be the WAN for a gluonized airGateway device (e.g. because it's an endpoint of a WDS link).
Ideally this should be configurable in config mode (as Matthias says: #284). And since config mode always listens on LAN we're safe to plug in our laptops even in the current setting.

commit b45585a1 (2017-02-23): Switch LAN and WAN port of Ubiquiti AirGateway

This makes the LAN port the PoE master, so the AirGateway can be used to
supply power to another node connected via mesh-on-LAN.

Fixes #608
closed 2017-02-23 by NeoRaider