freifunk-gluon/gluon #609: New status page station display is not working with VLANs ontop IBSS mesh interface

lcb01a, 2015-12-30:
Hi everyone,

While testing the master build I've noticed that the station display is not working correctly when  using VLANs.
No stations on ibss0.* will be displayed.

This is due to the fact that the main Interface has to be used to get the stations. 
Manually starting /lib/gluon/status-page/providers/stations with ibss0 as parameter gives the correct result.

I suppose /lib/gluon/status-page/www/cgi-bin/dyn/stations should simply cut any trailing ".*" from the VLAN interface.

Cheers
Linus

NeoRaider, 2016-01-01:
@lcb01a, can you test if merging #597 fixes this?

lcb01a, 2016-01-06:
I will test this tonight and give you feedback

lcb01a, 2016-01-08:
I've tested the code and it is still not working correctly.
I'm getting a lot of java script errors regarding "Uncaught TypeError: Cannot convert undefined or null to object" and the graph is not showing any neighbors.
If you like to have a look at the status page:
http://[2a03:2260:3004:25:a2f3:c1ff:fe82:4d0a]/

NeoRaider, 2016-01-08:
@lcb01a, I've just pushed 0880159eb5bc279421ebceeb63f3e81b55820c5f which fixes the frontend issue. So all that is left is handling the remaining issues in #597.

closed 2016-02-05 by NeoRaider