freifunk-gluon/gluon #619: gluon-neighbour-info can't query neighbours over Mesh-on-WAN

NeoRaider, 2016-01-12:
Maybe a firewall issue. Reported on IRC.

NeoRaider, 2016-01-13:
The issue is that the requests are multicast, not unicast, and thus the replies don't match the RELATED firewall rules.

A quick workaround would be to generally accept UDP packets from port 1001 to high ports (1024+) on the WAN port from fe80::/64.

jplitza, 2016-02-02:
Why, again, are me more paranoid on the WAN interface (default: reject) than on the Freifunk interface (default: accept)? Not only does the latter mostly have an unfiltered, public IPv6 address in most Freifunk communities nowadays, but also is the WAN interface oftentimes behind a NAT router without a public address and thus better protected.

NeoRaider, 2016-02-02:
@jplitza, some communities had issues with DNS amplification attacks abusing nodes which had global addresses (e.g. directly connected to a cable modem), after which we installed the restrictive WAN policy...

I'll see if I can find time to write a patch for this bug tonight.

jplitza, 2016-02-02:
Wouldn't blocking port 53 be sufficient to prevent DNS amplification attacks? That's what we do on the Freifunk interface as well, which – as I said – has a public address, too.

But don't get me wrong, I'm not advocating to loosen the policy on the WAN interface, I'm suggesting to tighten the policy on the Freifunk interface.

And if the firewall rule you mentioned above (allow 1001→ephemeral port) is the intended solution, I can put that together too.

NeoRaider, 2016-02-02:
Yes, that's the solution I indented to implement.

PR #636 (2016-02-03): gluon-neighbour-info: Open firewall on WAN for respondd replies
If the query was via multicast, the response isn't matched by
--state ESTABLISHED,RELATED

This fixes #619
commit 5bb4316f (2016-02-03): gluon-neighbour-info: Open firewall on WAN for respondd replies

If the query was via multicast, the response isn't matched by
--state ESTABLISHED,RELATED

This fixes #619
closed 2016-02-03 by NeoRaider