freifunk-gluon/gluon #624: wie wäre es mit neuem branch als refferenz

viisauksena, 2016-01-17:
sowas wie 2016.1 ?  oder der verpasste 2015.2
auf den man sich beziehen kann, oder wird es soetwas in bälde geben

NeoRaider, 2016-01-17:
Wir werden einen  Tag v2016.1 und einen Branch v2016.1.x anlegen, sobald wir dieses releasen (was wir Ende Januar tun wollen), bis dahin gibt es diese Versionsnummern nicht.

closed 2016-01-17 by NeoRaider

viisauksena, 2016-01-17:
nice to know ... weil auf den docs seiten gibts ja zum beispiel 2015.1.2 und dann dachte ich sollte es doch irgendwie auch einen neueren branch hier geben , 
danke für Aufklärung !
http://gluon.readthedocs.org/en/v2015.1.2/

NeoRaider, 2016-01-17:
v2015.1.2 ist ein Bugfix-Release aus dem Branch v2015.1.x, der zu v2015.1 vom master abgespalten wurde. Genauso wird es ab v2016.1 einen Branch v2016.1.x geben, und dann Bugfix-Releases v2016.1.1, v2016.1.2, ...

rotanid, 2016-01-17:
ja, gibt es:
https://gluon.readthedocs.org/en/latest/
solange kein Release erfolgt ist, gibt es keine neuere Versionsnummer als v2015.1.2, nur "latest", dessen Inhalt sich dauernd ändert

FreifunkUFO, 2016-02-09:
wieso ist eigentlich derzeit nicht der master gleich dem 2016-1? 
https://github.com/freifunk-gluon/gluon/compare/v2016.1.x...master

NeoRaider, 2016-02-09:
@FreifunkUFO, v2016.1 ist released, master ist Entwicklung für v2016.2.