freifunk-gluon/gluon #626: uhttpd is not listening to link local address

mmalte, 2016-01-18:
I am trying to reach the status page via the link local address to use a fe80 IPv6 as next node address.

uhttpd is configured to listen on:

> list listen_http '[::]:80'

The dropbear ssh server is listening to the fe80 addresses on br-client and local-node@br-client, so I can be sure both of them are reachable from my client.

This issue applies to the traditional an the new status page.

NeoRaider, 2016-01-18:
I don't think any common HTTP clients even support link-local addresses (AFAIK at least no browser does, with good reasons). I don't see why the server wouldn't handle requests on link-local addresses, so this is most likely a client issue.

Please note that URLs should be more or less globally valid (or at least inside a whole "intranet" or so), so link-local addresses containing interface names aren't supposed to work.

closed 2016-01-18 by NeoRaider

bittorf, 2016-01-18:
@NeoRaider it's a pity that most browser dont support linklocal addresses. i dont think there are "good reasons" for that. curl works for and some more can work with it, see https://forums.he.net/index.php?topic=3058.0
(i will leave the ticket closed, this is a client issue - uhttpd works with it)

mmalte, 2016-01-18:
Yeah, you are right. Access via forward to localhost is possible.

What a pitty, would have been great to this as a next-node-address for multible Communities.