freifunk-gluon/gluon #629: /lib/gluon/release > Invalid image, hardware ID mismatch, hw:08410010 image:08410009.

viisauksena, 2016-01-24:
in der /lib/gluon/release steht für die 841 der v9 
wenn der das dann als basis für den updateprozess nimmt failed das.

Invalid image, hardware ID mismatch, hw:08410010 image:08410009.
Image check 'platform_check_image' failed.

(kann auch sein wir haben was verbockt, ist jetzt nur an 2 routern aufgefallen die config files gesaved haben)

NeoRaider, 2016-01-24:
Die /lib/gluon/release enthält nur die Versionsnummer, diese wird direkt aus der site.mk genommen. Mit dem Modell hat diese Datei nichts zu tun.

Welches Modell ein Router hat, wird vom Autoupdater auf ar71xx anhand der /tmp/sysinfo/model bestimmt, das Ergebnis lässt sich z.B. durch den Befehl

    lua -e 'print(require("platform_info").get_image_name())'

anzeigen.

Der Autoupdater sucht dann im Manifest nach der Zeile für das passende Gerät und lädt das referenzierte Image herunter. Ich vermute, ihr habt irgendwie ein kaputtes Manifest erzeugt, das für den 841v10 ein Image für den 841v9 referenziert.

NeoRaider, 2016-01-24:
Oh, ich habe deinen Bug-Report falsch gelesen, ich dachte, es wäre der Autoupdater beteiligt. Dann kannst du alles bis auf den ersten Absatz meiner Antwort ignorieren.

closed 2016-01-24 by NeoRaider

viisauksena, 2016-01-24:
oh man, mein fehler ... ich bezog mich auf /tmp/sysinfo/board_name und da steht die falsche Bezeichnung drin, gleiches gilt für den cpe210 da kommt aus diesem cpe510

viisauksena, 2016-01-24:
@NeoRaider 
und hab das gleich nochmal validiert 

    # echo $(cat /tmp/sysinfo/model) $(cat /tmp/sysinfo/board_name
    fdf0:9bb:7814:a630:62e3:27ff:fe5a:1fe4 done
    TP-Link TL-WR841N/ND v10 tl-wr841n-v9
    fdf0:9bb:7814:a630:62e3:27ff:feee:458 done
    TP-Link TL-WR841N/ND v10 tl-wr841n-v9

wenn die daten aus dem model kommen kein problem, wenn aus board_name dann schon

viisauksena, 2016-01-24:
reopne, mein fehler meinte nicht die release sonder die /board/name ...
siehe dortige comments

NeoRaider, 2016-01-24:
@viisauksena, ja, das ist korrekt. board_name enthält denselben Bezeicher für alle Geräte mit kompatiblen Chipsätzen, das kann teilweise ein völlig anderes Gerät sein (normalerweise das erste Modell, bei dem ein neues Board vorkommt, so teilen z.B. sich insgesamt 7 Geräte das Board "tl-wr741nd", darunter der 841v5 und der 941v4). model enthält den Modell-spezifischen Namen.

viisauksena, 2016-01-24:
und nur zur Sicherheit gefragt, der string wird im gesamten updateprozess nicht benötigt ..

NeoRaider, 2016-01-24:
Der String ist primär für die initiale Konfiguration der Netwerk-Interfaces, LEDs u.ä. verantwortlich. Beim Update wird er nicht verwendet.