freifunk-gluon/gluon #635: MoW enabled forced nodes to have LAN connected during boot

Adorfer, 2016-01-31:
If you enable MoW (network.mesh_wan.auto=1) 
this at 841v8/v9/v10 will NOT work properly 
(this may apply to more ar71xx and other targets, but untested by me)

consequences 
a) if no LAN (correct, i do not mean WAN) eth link established during boot time, the node will not be reachable by it's IPv6 (from the site.conf's prefix). -> no working bat0-ipv6 (announced, ssh, icpm-echo)
b) probably (not fully tested): if LAN established during boot phasis, IPv6 is present, but ping to the IPv6 causes DUP-responses.

having WAN link established during boot does not make a difference.

uci show, ip a of node in status a)
http://pastebin.com/bwdv58Lg

kampa133, 2016-02-10:
Is wifi-mesh active in this scenario?

Adorfer, 2016-02-10:
@kampa133 
Wifimesh is active. 
Node is actually running when affeced, but not reachable via IPv6 from site conf's prefix. (but via fe80 hardware addresses)

NeoRaider, 2016-02-10:
The log in the pastebin doesn't match that symptoms. In the log, br-client isn't even up, so the node should not be reachable at all...

NeoRaider, 2016-02-10:
I suspect that this is a netifd bug. I'll have a closer look at it soon.

Adorfer, 2016-02-17:
on a build based out of 2016.1 i could not reproduce the issue.

NeoRaider, 2016-02-17:
@Adorfer, on what version did this issue occur then?

Adorfer, 2016-02-17:
@NeoRaider the build was from gluon-v2015.1-336-g4ce85af

NeoRaider, 2016-02-17:
Weird, there should not be a significant difference between these versions.

jannic, 2016-02-26:
Just to add a few data points, even though I'm not sure what is happening:

During testing of 2016.1 for the Aachen community, I had two cases where a 841n v9 became unreachable. These where 2 out of ~10 upgrade tests to that firmware I did yesterday on 841v9/v10 hardware. And it where exactly the two routers where mesh on wan is active.
(Both of them where running an older build of 2016.1 before, where the only change was the 5GHz bssid, so not relevant to 841 routers at all)

One of them, actually has a LAN port in use, but only connected to a switch, where no client was connected at that time. Connecting a client later had no effect, but I could observe that the Freifunk client network was available and usable on that port. Additionally, the router still forwarded mesh traffic via WLAN. So it looked like only the IP connectivity directed to the router itself was broken.
(I saw a lot of duplicates during an ICMP ping which went through a batman link via that router, though. Not sure if that was related.)
It was partly possible to connect to the router on the LAN cable through it's link local address: It was reacting on port 22, and the ssh handshare started successfully, but the router didn't accept my ssh key. (I only have public key authentication activated, so password authentication was impossible, as well).
But: After trying to reboot the router, it didn't come up again, instead all LEDs where flashing every few seconds, and the only way I found to revive it was to flash a new firmware via TFTP.
So it may be that it was just a case of a failed firmware upgrade, for whatever reason.
After that, it started to work, without changes to the cabeling, ie. with mesh on WAN activated and a single LAN port connected to a switch.

The second affected router is still unreachable, it's in the basement of a different building and I didn't have a look at it, yet. So if you have any idea what I could try to debug it before rebooting, please tell me.

Then I tried to reproduce the situation on a third router where I have a serial console attached. It's also an 841v9. Unfortunately(?) I was unable to reproduce an identical situation. What I can reproduce is a situation where it seems to be to handle batman traffic normally, does have an IPv6 address (visible from the serial console), but doesn't answer to any incoming connection. But by now, this situation was always reversible: It was enough to connect a wireless client (so not even a LAN cable was needed), and IPv6 started to work immediately. In some cases, it also started to work spontaneously after ~2 minutes.

I'm running out of ideas what to try next - any hints welcome.

viisauksena, 2016-03-20:
i can not reproduce the error : here some postitve testing : https://forum.freifunk.net/t/neue-beta-version-2016-1-2/11345/8?u=fuzzle

viisauksena, 2016-03-21:
@Adorfer .. i am hunting another bug/ maybe bug (20 % of last update cycle nodes did not come back :(  )
so i suspicous what you mean with .. MoW **need** LAN ... what exactly is the behaviour than - and is it fixed by itself when plugin ethernet or turn MoW off?
could it simply be that these nodes were in config mode, at least for one short iteration?

NeoRaider, 2016-03-21:
@viisauksena, it seems more likely that the offline nodes are affected by #669, which is not fixed after all...

NeoRaider, 2016-05-30:
Closing, this problem seems to be gone...

closed 2016-05-30 by NeoRaider