freifunk-gluon/gluon #637: Amplification attacks via respondd

jplitza, 2016-02-02:
Currently, respondd replies to tiny (<50 bytes) UDP queries from global IP addresses with a much larger chunk of data (at least up to around 1 KB). For example, try this:
```
gluon-neighbour-info -d 2a06:8782:ffbb:1337:12fe:edff:fee5:fb22 -r 'GET nodeinfo statistics neighbours' -p 1001 -t 1
```
This is the ideal scenario for amplification attacks. A first step could be to only allow link-local addresses to query respondd, as is currently already done on the WAN interface. Question is, are there any valid use cases where a non-link-local address will query a respondd, and are they important enough to support them.

Adorfer, 2016-02-03:
i do not see any use valid scenario for external access for respondd requests. 
(if some comes, we should then think of some kind of usage-throtteling for incoming requests on the nodes.)

viisauksena, 2016-02-05:
good design include ideas like @jplitza - i wouldlove to see a protection, maybe put a note that this must not be restricted if you want to compile a version for a legitimate "other"usecase

NeoRaider, 2016-02-05:
I agree that we should block requests from non-LL addresses for now. It's better to be safe than sorry, and if a new usecase comes up, we can still loosen the policy later.

PR #639 (2016-02-05): gluon-respondd: restrict queries from the mesh to link-local addresses
Restrict to link-local addresses to prevent amplification attacks from
outside the mesh, or such attacks affecting the outside world.

Fixes #637
commit 2f499dbf (2016-02-05): gluon-respondd: restrict queries from the mesh to link-local addresses

Restrict to link-local addresses to prevent amplification attacks from
outside the mesh, or such attacks affecting the outside world.

Fixes #637
closed 2016-02-05 by NeoRaider