freifunk-gluon/gluon #641: gluon-luci-wifi-config: txpowerlist is likely incomplete/wrong

Adorfer, 2016-02-06:
to obtain a valid "txpowerlist" from iwinfo in CC, you need to have the radio interface fully initialized (including regdomain and radio channel). 
This is currently not the case. 
by consequence the maximum power is for 841v9 at 19dB, and 841v10 at 18dB. 
In case a user selects the highest available (18dB) and saves that value, he is stuck with a txpower of 18, even if later (during full operation of the node) the txpowerlist would go up to 20 (unless you are running on ch13...)

NeoRaider, 2016-02-06:
This may be true, but the txpowerlist doesn't really mean a lot anyways... even when the txpowerlist contains higher entries, the power actually used (shown by iwinfo) is limited to 16dBm on 841 v9 and v10.

viisauksena, 2016-02-06:
this is also heavily discussed here : https://github.com/freifunk-gluon/gluon/issues/555

Adorfer, 2016-02-06:
@viisauksena #555 is about the actual radio performance on AT9k  (claim: has changed from OpenWRT BB->CC)
If that is the case #555 has to be fixed upstream (or considered as "won't fix": upstream). 

This  issue hiere (#641) is about the LUCI-Wificonfig, which uses a non useful strategy to obtain the available txpowerlist. This can (and should) be fixed gluon-luci-wifi-config.

The current behaviour of _"iwinfo <dev> txpowerlist"_ to provide just the regulatory available settings is correct (from my point of view). 
In other words: If we want a correct list, we have to initialize the radio with the regdom, channel and htmode, in order to obtain a list for selection which reflects what is considered as avalid (by the manufacturer). 
If we do not care about a correct list (and take whatever output is coming).

@NeoRaider in case the output of the list is not correct (wrong values in ART tables), this would be not an issue of Gluon. But at least we should try to obtain what is available.

NeoRaider, 2016-02-06:
@Adorfer, I don't know how you got 20dBm in any list, I've never seen anything higher than 16dBm with current versions on DE regdom.

Adorfer, 2016-02-06:
841v10: 
```
root@v10testexp:~# iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
root@v10testexp:~# iwinfo radio0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
  16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)

```

RubenKelevra, 2016-09-09:
@NeoRaider I think this bug can be closed as well, right?

Adorfer, 2016-09-29:
is the WiFi now initialized (to final radio channel and HTmode) before obtaining the txpower list? 
(I am talking about config mode, not txpower list after normal boot up)

NeoRaider, 2017-02-23:
We can't initialize the WLAN in the setup mode, so this can't be improved at the moment.

closed 2017-02-23 by NeoRaider