freifunk-gluon/gluon #643: Ads, Spyware, Tracking

waschmi, 2016-02-06:
Blocking Stuff is always a sensitive subject but since Freifunk (at least some Citys) kick out Nodes with Clients using Torrents then blocking ads seems okay too. It will reduce 'useless' traffic and will give more bandwidth for useful things. Doing it via Hostfile seems doable. OpenWRT might already have some adblocking mechnism in place.

Hosfile example: http://winhelp2002.mvps.org/hosts.txt

jplitza, 2016-02-06:
I don't even want to start the ethical discussion, and simply say that if this was to be done, it would have to happen on the gateways, not on the nodes. (Also, that list that you linked is even with gzip compression still 128 KB in size. On a node with 4 MB flash, that is unacceptable.)

closed 2016-02-06 by jplitza

Adorfer, 2016-02-20:
I fully support jplitza. the node is the wrong place. 
And Freifunk is PPA. there is not much room for discussion on this.