freifunk-gluon/gluon #644: request: transmit via collectd additional node data: wifi channel(s)

Adorfer, 2016-02-07:
i'd like to have transmitted via the collectd service (gluon collector) more node static data, in order to monitor remotely the setup of more complex mesh networks without the need to ssh to nodes to check for data.

- radioX.wifichannel selected.
- radioX.ClientAP disabled
- radioX.wifimesh disabled 

perhaps 
- meshvpn /disabled/disconnected/connected
- MoW enabled
- MoL enabled

ultimativly even, if info available from switch
- ethernetports up

e.g. where this would help a great deal: https://map.eulenfunk.de/#!v:g;n:60e327603e7c

plumpudding, 2016-02-07:
we have meshvpn already
MoW and MoL can probably be extracted from the current data too (only if one of the two is active, not which)

NeoRaider, 2016-02-07:
I think you mean respondd, not collectd, right? We don't support collectd.

NeoRaider, 2016-02-07:
We already provide a list of mesh interfaces with type (wireless/tunnel/other, where other is usually wired).

I guess we could add:
* Channel of each wireless mesh interface
* List of wireless client interfaces (with channel)

I think we shouldn't add anything hardware-specific, like refering to specific radio devices or specific ethernet ports, or differentiating between LAN and WAN ports.

plumpudding, 2016-02-07:
> I think you mean respondd, not collectd, right? We don't support collectd.

He wants to collect this data with gluon-collector from ffdo/node-informant
So it has to be provided by respondd, yes.

Adorfer, 2016-02-07:
@NeoRaider 
> I think you mean respondd, not collectd, right? We don't support collectd.
Correct, my fault. I took the "gluon collector" for collectd... but i off course it's (c)-respondd

tcatm, 2016-02-08:
> fastd is up, but not batman originators on the other side

The neighbour information provides just that. Please keep in mind that it is possible to have batman originators on an interface that you aren't made visible by the kernel module (due to lower TQ mostly).

viisauksena, 2016-06-17:
you maybe should have a look at https://github.com/viisauksena/gluon-airtime 
there - for now in progress - are channels spit into respondd and alfred for 2g and some more info (related to airtime)

frennkie, 2019-09-07:
is there any progress on this?

rotanid, 2019-09-07:
> is there any progress on this?

the label "help appreciated" means, that someone has to step up and start working on it :)

rotanid, 2024-12-23:
some (all?) of it has already been done. havin this 8 year old issue doesnt help with additional infos.

closed 2024-12-23 by rotanid