freifunk-gluon/gluon #647: Gluon 2016.1.x needs 35GByte disc space

Byggvir, 2016-02-10:
Hello

I generated a new Gluon from 2016.1.x. It used 35 GByte disc space.

> du -h -d 1
823M    ./output
3,5M    ./package
296K    ./docs
16K     ./contrib
29G     ./openwrt
75M     ./packages
48K     ./scripts
24K     ./include
5,6G    ./build
1,2M    ./patches
116K    ./targets
4,4M    ./.git
268K    ./site
35G     .
> cd openwrt
openwrt>du -h -d 1
19M     ./package
35M     ./target
132K    ./docs
417M    ./dl
1,5M    ./scripts
1,3M    ./toolchain
304K    ./include
2,6G    ./staging_dir
44K     ./config
15M     ./.git
24M     ./tmp
26G     ./build_dir
44K     ./logs
2,8M    ./tools
29G 
openwrt/build_dir> du -h -d 1
1,2G    ./host
2,5G    ./target-i386_i486_uClibc-0.9.33.2_gluon-x86-generic
2,5G    ./target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-nand
2,4G    ./target-i386_i486_uClibc-0.9.33.2_gluon-x86-kvm_guest
4,0G    ./toolchain-powerpc_8540_gcc-4.8-linaro_uClibc-0.9.33.2
2,5G    ./target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic
2,5G    ./target-powerpc_8540_uClibc-0.9.33.2_gluon-mpc85xx-generic
3,8G    ./toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2
4,0G    ./toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2
26G

Is it neccassary to keep the build_dir targets for an next run?

jplitza, 2016-02-10:
It isn't *neccessary*, but building the toolchain amounts for a significant portion of the total time the build takes. You can remove most of the intermediate byproducts by running `make dirclean` in the root of Gluon.

NeoRaider, 2016-02-10:
When using `make dirclean`, you should take care to save the opkg repo signing key (see https://gluon.readthedocs.org/en/v2016.1/user/getting_started.html#opkg-repositories ).

Otherwise, as @jplitza wrote, you can clean your tree as often as you want, but at least similar targets (for example ar71xx-generic and -nand) will benefit from each other as they share the same toolchain. Building the same target twice (with an updated site.conf or site.mk for example) will work a lot faster if you don't clean your tree between builds.

closed 2016-02-10 by NeoRaider