freifunk-gluon/gluon #648: Support for Mediatek/Ralink WLAN

NeoRaider, 2016-02-10:
Ralink WLAN hardware does support VIFs, but
* no VIFs with IBSS (but AP + 11s works)
* only with sequential MAC addresses (the MAC addresses of all VIFs may only differ in the last 3 bits -> 8 VIFs are supported)

The MAC address scheme deployed by Gluon doesn't not generate sequential MAC addresses (the address of a single radio are counted in byte 2, different radios in byte 3). This was done to avoid address collisions (we have to deal with devices with only a single assigned MAC address, so devices bought together often have very similar MAC addresses).

We plan to fix this by generating a new base MAC address by applying a hash function to the real primary address, and derive the VIF addresses from this base address.

Open questions:
* Use such hashed addresses for all kinds of addresses currently generated by generate_mac(), also ethernet?
* Generate a new hash with different input for each radio, or derive different MAC address ranges for each radio from a single hash? Using a hash for each radio will lead to very different addresses to be assigned to different interfaces of a single node. We should evaluate which scheme yields the least probability of collisions.

FreifunkUFO, 2016-02-11:
your expectations are wrong: at least some mediatek/ralink hardware support ad-hoc + vAP mode.

p.s. for the hardware which doesnt support that: gluon already has queckboxes for disabling mesh or vAP on the device!?

NeoRaider, 2016-02-11:
I've updated the issue description with a more detailed plan.

@FreifunkUFO: That's good to know, but doesn't really change a lot. The missing IBSS support in some models is reason enough to switch to 11s as soon as possible, so we plan to generally phase out IBSS support in Gluon as soon as we have a migration plan.

FreifunkUFO, 2016-02-12:
so here a small ticket to support at least devices, wo can "ad-hoc + vAP":
https://github.com/freifunk-gluon/gluon/issues/650

waschmi, 2016-02-13:
Archer C50 (mediatek mt7620a and mt7612e)
For Gluon I had to make MAC entries manualy. IBSS works but not IBSS+AP. 11s I didn't test, yet. 
[openwrt-archer-c50.TXT](https://github.com/freifunk-gluon/gluon/files/129237/openwrt-archer-c50.TXT)
(reset button doesn't work. firmware update over config-mode page works ok.)

jotzt, 2016-04-16:
D-Link DIR-860L (`ramips-mt7621`) is sucessfully running with L2TP/802.11s under Gluon v2016.1.2. 
 - target was added according to the Gluon documentation
 - 11s is stable but slow, only MCS0 (6.5 Mbit/s) at 2.4G and 5G so far
 - client connection is barely stable
 - correct primary MAC-address can be set similar to TL-WDR3600 `/sys/class/ieee80211/phy1/macaddress` 

Anything I could test on that DIR-860L in order to support the development?

Status: http://karte.freifunk-muensterland.org/map07/#!v:m;n:e46f132ebcac

commit c73a12e0 (2016-04-27): Change MAC schema generation (#715)

While ath9k/ath10k devices can supprt VIFs with any combination of MAC addresses, there are also adapters which have a hardware MAC filter which only allows a few bits to differ. This commit changes the addresses of all VIFs to ony differ in the last 3 bits, which is required to support many Ralink/Mediatek based WLAN adapters.

Technically, the new addresses are generated by calculating an MD5 hash of the primary MAC address and using a part of this hash as a prefix for the MAC addresses.

The addresses (BSSIDs) of the AP VIFs are also reused for the LAN a [... cut]
closed 2016-04-27 by NeoRaider