freifunk-gluon/gluon #649: Support for Openwrt  Imagebuilder and SDK

FreifunkUFO, 2016-02-11:
on openwrt special changes or more packages are usually done with Imagebuilder / SDK.

Gluon as an openwrt-project should provide Imagebuilder and SDK!

belzebub40k, 2016-04-20:
Would be very handy to have the imagebuilder as we build images for 3 communities with the same codebase and the main differences in site.conf. Imagebuilder would save us a lot of time and would make sure that binaries for all communities are identical.

Did you already made some progress in adding imagebuilder to Gluon?

NeoRaider, 2016-04-20:
We probably won't have a distinct image builder with Gluon in the near future. A Gluon build will take just a few minutes - about the time the OpenWrt image builder would take - if a fully built Gluon tree is re-built without cleaning it inbetween.

To build multiple sites, just set GLUON_SITEDIR and GLUON_OUTPUTDIR to three different directories and re-run `make` for the desired targets.

As a general rule: when doing changes to site.conf or site.mk, a `make clean` is never necessary; this is also true for changing the GLUON_SITEDIR (unless the sites have different *modules* files). You only need to clean after a `git pull`/`make update`.

belzebub40k, 2016-04-20:
Thanks for the hint. Wasn't aware of that but sounds reasonable.

FreifunkUFO, 2016-05-23:
the compiling time is NOT important. imagebuilder is for using openwrt in a normal "openwrt" way. 

the SDK is even more important!
its needed by almost everyone who wants to build own packages, other normal ("gluon-out-of-scope") openwrt packages and for rebuilding (updated) openwrt packages in the future!

p.s. did you read about that airos-virus? AirOS/UBNT didnt publish their SDK any more, so people are forced to use outdated firmware with that old SDK.

NeoRaider, 2016-05-23:
I agree that the SDK would be nice (although I've never used the OpenWrt SDK, but always use a normal OpenWrt tree for package builds). Can the SDK build kernel modules?

For userspace packages, a "Gluon" SDK is not needed at all, you can just use an OpenWrt CC SDK.

FreifunkUFO, 2016-05-26:
yes, SDK is for builing/compiling modules and packages.
each gluon (f.i. download folder for a specific town) should contain IB+SDK!
openwrt-cc-sdk cant be used, because gluon is mostly openwrt-cc-trunk or openwrt-bb

closed 2025-04-04 by neocturne