freifunk-gluon/gluon #650: Support for Mediatek/Ralink Platform

FreifunkUFO, 2016-02-12:
Mediatek platform is nice and already supported by openwrt. 
Please support at least some devices, which supports ad-hoc + vAP mode.

For devices "no VIFs with IBSS" please see at https://github.com/freifunk-gluon/gluon/issues/648

NeoRaider, 2016-02-12:
Why do we need another ticket? Please keep this in #648.

closed 2016-02-12 by NeoRaider

FreifunkUFO, 2016-02-15:
its not a duplicate! 
its for all mediatek devices, which should be supported out of the box.
so you dont need to plan all these stuff mentioned in that other ticket (when ad-hoc AND vAP is NOT supported)

NeoRaider, 2016-02-15:
@FreifunkUFO, the main point of the other ticket is *not* the missing AP+IBSS function. #648 is about any type of Mediatek support.

If the devices which support AP+IBSS can also work with any MAC address combinations, that will make supporting them easier, but we still don't need another ticket for them.

FreifunkUFO, 2016-04-01:
main point at other ticket still seems "missing AP+IBSS" (see text!)
so we should use another ticket for that

FreifunkUFO, 2016-07-01:
neoraider seems to want to test this platform now:
https://github.com/freifunk-gluon/gluon/commit/f1570cacb0f61c564505fa32124a8fe6aec4ae3a