freifunk-gluon/gluon #653: Remove contact Data in new Status Page

mmalte, 2016-02-15:
Publish the contact information worldwide will prevent a lot of users from entering contact information at all.

It should be possible so select whether to publish this information or not.

Choosing this in the config mode would be great, choosing it in the site.conf would be sufficient.

LeSpocky, 2016-02-15:
What's the difference to publishing those data to the world via alfred or respondd?

mmalte, 2016-02-15:
There is a valid global IPv6 Link on our Map. Any Crawler with IPv6 is able to follow this link.

I guess you agree that publishing via alfred/respondd to the local mesh is something totally different.

LeSpocky, 2016-02-15:
Well this is only true for communities with valid global IPv6 addresses …

NeoRaider, 2016-02-15:
The basic idea of this config mode field is to have contact information available to everyone. If someone doesn't want to have their information published, they should not supply it there. If you don't want users to publish the contact info, remove or replace the *gluon-config-mode-contact-info* package. It says there *in bold letters* that the info is public.

The information published via alfred and respondd is public, and any user in the mesh could use and republish it. The status page API is simply a mapping of the repondd data to HTTP. This makes this a very simple choice: either send it via HTTP as well, or not at all. Filtering data while converting between protocols doesn't seem reasonable to me.

By the way, is Google able to parse the JavaScript and extract the JSON data that is fetched in the scripts? If not, the contact info won't be visible on Google anyways.

rotanid, 2016-02-15:
schließe mich da NeoRaider an.
Ist natürlich problematisch, wenn man als FF-Community bisher etwas zugesichert hat, das man bisher auch schon nicht halten konnte (Kontaktdaten geheim halten) - aber das ist ja nicht das Problem von gluon.

viisauksena, 2016-02-16:
man könnte aber die email verhexeln .. so dass sie nicht mehr so leicht maschinenlesbar ist ... 
mögl. 1 : wirkliches verschlüsseln der kontaktdaten (zum beispielmit fastdkeys der gateways) und diese dann nur als solches über alfred und dergleichen propagieren ... da würde dann im meshviewer bspw. nur nen string rauspurzeln ... und man müsste im Hintergrund was bauen das Menschen diese Kontakte benutzen könnten - für admins mit zugang zu den fastd secret keys kein problem - für allerwelt müsste dann eine art contact formular her... (schwierig und macht evtl. 100 neue Fässer auf, aber chic)
mögl. 2 : zumind. das bla@foo.bar zu ner bla _l@l_ foo bar machen im meshviewer selbst

mmalte, 2016-02-16:
Möglichkeit 1 gefällt mir sehr gut, wir haben auch schon darüber diskutiert wie wir die Kontaktdaten sinnvoll verschlüsselt übertragen könnten, die wirklich charmante Idee die bereits vorhandenen fastd Schlüssel zu nutzen ist uns bisher nicht gekommen.
Die Frage ist nur welche Software ist fürs Verschlüsseln einer Datei mittels fastd notwendig?

Schön wäre dann eine Wahlmöglichkeit auf der Configseite:

- Kontaktdaten auf Karte darstellen
- Kontaktdaten nur an die Admins übertragen

viisauksena, 2016-02-16:
weis ich jetzt auch nicht .. mit  uci get fastd.mesh_vpn_backbone_peer_sn1.key bekommt man nen pubkey, wie man den aber jetzt benutzt im etwas von stdin auf den kisten zu verschlüsseln???

NeoRaider, 2016-02-16:
Please note that this bug tracker's language is *English*. You may use German on the mailing list or on IRC.

Regarding the public keys used for fastd and the autoupdater: In theory, it would be possible to implement ElGamal or a similar asymmetric encryption scheme based on libuecc. If anybody does that (I certainly won't), using a separate keypair configured in site.conf would make sense.

There is one more thing I'd like to point out though: respondd is designed to let any user get the same information. Any user can setup their own instance of the mesh viewer or a visualization tool of their choice. You shouldn't need to be part of a group of "trusted administrators" to contribute to the network in this way. I don't see why any adminstrator should have the right to get information a normal user can't.

mmalte, 2016-02-16:
Even within the Freifunk network there are certain things which are only allowed to these "trusted administrators", critical things like the Autoupdater. Personal Information like the IP Address of the Uplink DSL.

In my opinion the contact data is personal information and the user should choose whether it should be published or not.
If you change the default behavior you should always set the option that favors privacy as default.

We'll look into possibilities to add this to the config mode.

NeoRaider, 2016-02-16:
This was not a change of default behaviour, the information was always public (and many networks display this information in the mesh viewer). We can't change the definition of this field just because you misinterpreted it.

But you can define your own field, extend the config mode to support your field, and add a way for administrators to get this information. We might even accept a pull request if the implementation is technically sound.

closed 2016-02-24 by tcatm