freifunk-gluon/gluon #656: update-fallback module if network seems faulty (from node perspective) wanted?

viisauksena, 2016-02-18:
general question if a gluon-update-fallback module is wanted?

this module would check if **batctl gwl -N** see some GW, and by that assuming that its batman network could be faulty.
This could be due channel change, l2tp<->fastd, fastd_mtu, batv14_v15, changed sn_keys, missed updates for a long time because of being offline etc.
and then login as client in his well known ssid and try to do a **autopdate -f**

see also discussion here: https://forum.freifunk.net/t/batman-v14-zu-v15-upgrade-gluon-package/
see example here : https://github.com/viisauksena/gluon-v14tov15-helper

if this is wanted, i make a cleanup and propose it as pull request

jplitza, 2016-02-18:
`batctl gwl` is a bad measure for "is everything alright". In particular, why not simply see if the update server is reachable? In general, yes, something like this is a longstanding wishlist item, see #427. But it would have to be more integrated in the update process, and shouldn't be done independently from the autoupdater in another cronjob.

I'm closing this issue, please keep discussions in #427.

closed 2016-02-18 by jplitza

viisauksena, 2016-02-18:
ah thx, forget about this issue 
(while the paket offert here is doing right that , with own cronjob check som "things" and than call autoupdater")

Adorfer, 2016-02-20:
As pointed out in the forum 3 days ago: i do not see the point in disturbing "offline clouds" by turning to client mode to fast, too often, since there is potential regression scenarios at least in some radios (where you can't do ibbs, ap and cli at the same time, especially not on other frequencies.)

https://forum.freifunk.net/t/batman-v14-zu-v15-upgrade-gluon-package/10808/5