freifunk-gluon/gluon #657: Apple Devices in Mesh Neighbor list

mmalte, 2016-02-18:
I got some weird Mesh Neighbors on the new Status page:
[image]

Are these Devices which are set to connect to our Mesh SSID?

NeoRaider, 2016-02-18:
Indeed, these are some random devices accidentially connected to the IBSS network. They are shown in light grey because they are inactive for more than 1s (60s in this case).

I don't really see a nice way to filter these...
* We can't check if these neighbours speak batman-adv as batman-adv doesn't provide us with a list of neighbours that aren't currently the best path for any destination
* We *might* limit the list to neighbours that respond to respondd requests, but that would also make the list less reliable (for example to align antennas)

If we really care, one way to solve this would be to provide a checkbox, so by default only "verified" neighbours are shown (we know it speaks batman-adv or respondd), and only after checking "show all neighbours" such weird entries would be shown.

Also, it will probably solve itself when Freifunk networks migrate from IBSS to 11s... until Apple devices learn 11s.

mmalte, 2016-02-18:
It is absolutely okay to show them as long as they are removed after some time, but I've seen them for quite some time with the counter stuck at 60s.

After I've closed the statuspage they are all gone. I'll leave this statuspage open for some time to see if it happens again.

NeoRaider, 2016-02-18:
Ah, I guess the status page is missing code to remove vanished neighbour entries.

tcatm, 2016-02-24:
If it happens again, could you post both the statuspage output, http://$node/cgi-bin/dyn/neighbours-batadv and http://$node/cgi-bin/dyn/stations?ibss0 without redacting any MACs?

NeoRaider, 2016-02-28:
@tcatm, this is easily reproducible, neighbours are never purged from the list (happens also with real neighbours that have been switched off).

tcatm, 2016-02-28:
I think code for purging neighbours is already in place. If someone could collect sample data with a node disappearing that should be purged, I can have a look at the code to see what's going on.

commit 79bc3a8d (2016-02-29): gluon-status-page: fix neighbour purging

notUpdated is a Set, to a for-in loop doesn't work.

Fixes #657
commit dfd5d7bd (2016-02-29): gluon-status-page: fix neighbour purging

notUpdated is a Set, so a for-in loop doesn't work.

Fixes #657
closed 2016-02-29 by NeoRaider