freifunk-gluon/gluon #658: D-Link DIR-505 LED and Button-Functionality missing

rubo77, 2016-02-21:
Please add some support for the LED at the D-Link DIR-505.

It would also be possible to add some behaviour to the Button on the side and the switch at the back.

thorough discussion available here:

https://forum.freifunk.net/t/d-link-dir-505-als-neuer-einsteiger-router/10759

NeoRaider, 2016-02-21:
Your report is missing essential information.

Which LEDs work, which don't?
Which buttons work, which don't?

How are these LEDs labeled, what do they do with the stock firmware?

rubo77, 2016-02-21:
Usually the LED is green when running, there is only one LED that can be red or green

[image]

http://www.dlink.com/-/media/Consumer_Products/DIR/DIR%20505L/Manual/DIR505LA1manual012513v10pdf.pdf

It already is red when in config mode, that's fine

viisauksena, 2016-02-22:
while booting with 2016.1 it starts with red (short) .. blinks green while booting and then is off (which is nice , but i would be ok if it blinks like the 841 while meshing)

NeoRaider, 2016-02-23:
Ah, OpenWrt has defined the green LED to be used both as status LED (so it blinks during boot) and for eth1. The network configuration is b0rked (eth0 is dead, eth1 is the ethernet port), which should really be fixed in OpenWrt (but will break updates when we fix it).

I don't like the LED being used for two different things, especially as it is used differently than in the stock firmware. The green LED should just be used as the status LED, nothing more. Defining different triggers for the same LED isn't really a good idea and may lead to funny results...

NeoRaider, 2016-02-23:
I've ordered a DIR-505, so I'll clean this up myself.

viisauksena, 2016-03-12:
isnt the topic misleading, as the abuse is in stock fw - they use LED to indicate boot things AND use it as WIFI indicaor .. and gluon / openwrt simply use the LED correctly as status LED .. and later its turned off. 

(or do i miss something)

NeoRaider, 2016-03-12:
@viisauksena, no. The LED is abused as ethernet indicator in Gluon/OpenWrt, so it is off when no ethernet cable is connected.

viisauksena, 2016-03-12:
oh ok, never used it like that - only meshing in remote areas .. so i missed that

commit 3d3eb24a (2016-07-28): ar71xx-generic: backport a few D-Link DIR-505 A1/A2 patches

Fixes #658
closed 2016-07-28 by NeoRaider