freifunk-gluon/gluon #663: mark target "ubiquiti nanostation" & "nanostation bulliet" as broken (for "factory" at least)

Adorfer, 2016-02-24:
Since AirOS5.6 fashing the gluon/openwrt "factory" will brick the device. 
This regularly happens users who did not read the instruction to downgrade to AirOS5.5.x before.

eriu, 2016-02-24:
UBNT Geräte sind nicht broken. passt doch einfach eure Anleitungen an :) (siehe https://wiki.md.freifunk.net/Anleitungen/router-flashen-ubnt)

Adorfer, 2016-02-24:
@eriu: Ich kenne die Anleitungen. 
Und ich kenne mehrere Leute, die die stable-Image "wie gewohnt" geflashed haben, wie schon viele vorher. -> Kazong!
Es würde fürs erste reichen, "factory" für diese Geräte nicht mehr zu bauen. Sollen die Leute, die es können nach dem Stock-Downgrade erstmal ein Openwrt-factory flashen und dann einen Freifunk-Gluon-Sysupgrade.

eriu, 2016-02-24:
@Adorfer 
Warum nicht Stock-Downgrade und dann normales Gluon-factory flashen. 

Warum ein Feature entfernen, nur weil Menschen die Anleitungen nicht lesen/anpassen? Erst OpenWRT und dann Gluon-sysupgrade zu flashen ist das gleiche wie direkt Gluon-factory einspielen. Auch mit OpenWRT kann man sich mit AirOS 5.6.x das Gerät bricken.

A-Kasper, 2016-02-24:
Am 24.02.2016 um 22:01 schrieb Adorfer:
> @eriu <https://github.com/eriu>: Ich kenne die Anleitungen.
> Und ich kenne mehrere Leute, die die stable-Image "wie gewohnt"
> geflashed haben, wie schon viele vorher. -> Kazong!
> Es würde fürs erste [...]

Ist mir auch passiert. Na und?! Man kann problemlos über tftp
zurückflashen und den Fehler korrigieren.

NeoRaider, 2016-02-24:
We should look into adding support for installations from the new stock firmware instead.

Adorfer, 2016-02-24:
@A-Kasper 
> Ist mir auch passiert. Na und?! Man kann problemlos über tftp.

Ich gehe davon aus, dass Dein Know-How das von durchschnittlichen Freifunk-Nutzenden übersteigt. 

(Und nein, ich kann kann auch mit JTAG auf den SOC, wenn der Bootloader auch nimmer will. Aber ich denke, dass der Anspruch von Gluon ein anderer ist. Nämlich der, mit einer "nonbroken" den Support für die marktüblichen Geräte zu gewährleisten. 

Wenn es nicht funktioniert, ohne dass vorher Doku genau liest und man dann den ftpd suchen darf, dann ist das schlicht BROKEN.

jjx-, 2016-02-24:
Man hat 10 Router erfolgreich geflasht und schaut dann nicht mehr in irgendwelche Anleitungen und dann passiert es halt. Und tftp kann nicht jeder.

A-Kasper, 2016-02-25:
Du willst also ein Gerät rauswerfen, weil es anders geflasht werden muss,
und Du es für unzumutbar hältst die Anleitung zu lesen?

Das macht für mich keinen Sinn. Die Firmware läuft stabil.

Das Einzige, dass ich wenn machen würde wäre ein image Unterordner "special
flashing procedure - read documentation" oder so. aber das kann eure
community selbst machen.

für mich gibt es einen großen Unterschied zwischen kaputter firmware und
bootloader downgrade notwendigkeit

NeoRaider, 2016-02-25:
I agree with @Adorfer here. If we can't support the NanoStation well, we shouldn't support it at all (meaning we mark it as BROKEN).

I've started looking into the bootloader differences. Fixing everything should be simple; so far I've found out that the "changed flash layout" information on https://wiki.openwrt.org/toh/ubiquiti/airmaxm is nonsense (the old and the new firmware use the same flash layout).

NeoRaider, 2016-02-25:
56f1cd5383b8cb44174f618dfb566c8fc9f2262f should fix this issue. It would be great if someone could verify that flashing from devices originally using AirOS 5.6.x is working fine now.

I've checked with a NanoStation Loco M2 which was originally using AirOS 5.5.x. Even when first updating to 5.6.x, OpenWrt was working fine; but I found various other issues during my experiments that should be fixed by this.

See also https://raw.githubusercontent.com/freifunk-gluon/gluon/56f1cd5383b8cb44174f618dfb566c8fc9f2262f/patches/openwrt/0042-firmware-utils-mkfwimage-fix-firmware_max_length-for-XM-layout.patch for some more information.

NeoRaider, 2016-02-25:
I might have found another issue; please don't test the fix yet, it might cause some more breakage...

NeoRaider, 2016-02-26:
I've fixed the remaining issue in 89b3dca8b5319657b37fa5ad736a65bef87f53e9, please test.

Adorfer, 2016-02-29:
@NeoRaider 
Zumindest für die XM (NS2LocoM) kann ich bestätigten: Scheint zu funktionieren von AirOS5.6 auf Gluon-Master (von gestern abend)
https://forum.freifunk.net/t/abgeschossene-rocket-m2-bzw-nanostation-neue-u-boot-version-von-ubiquity/6912/31

eriu, 2016-02-29:
Direct flashing from Nanostation M5 XW (AirOS 5.6.2) to gluon works with gluon master (#g79bc3a8) 
https://forum.freifunk.net/t/ubiquity-airos5-6-u-boot-aenderung-brick-gefahr-rocket-m2-bzw-nanostation-locom-erst-downgraden-auf-airos5-5/6912/77?u=eriu

NeoRaider, 2016-02-29:
Okay, great.

closed 2016-02-29 by NeoRaider