freifunk-gluon/gluon #664: Check if coordinate are valid

mmalte, 2016-02-24:
Would be nice to check if the geo ID is valid:

[image]

viisauksena, 2016-02-25:
but this should be better checked by the meshviewer .. something like, if distance >=25 km ignore

jplitza, 2016-02-25:
@viisauksena Distance of what? Why can't you have a community with a diameter greater 25km?
@mmalte Can you give what coordinates were actually specified for those two nodes?

viisauksena, 2016-02-25:
@jplitza i mean a single link between 2 nodes, could not imagine how you get 25km + with actual HW we use ..

tcatm, 2016-02-25:
We can't prevent nodes from announcing bogus information so this can not be fixed in Gluon. However, an improved coordinate input box could prevent mistakes.

Is that something you'd like to work on?

rotanid, 2016-08-22:
feel free to open a pull request if you want to do what @tcatm suggested.

closed 2016-08-22 by rotanid

viisauksena, 2016-08-24:
this question was for @mmalte .. right

rotanid, 2016-08-24:
for everyone.