freifunk-gluon/gluon #680: BananaPi M1 crashes with more than 2 Mesh-VLANs

belzebub40k, 2016-03-09:
When I configure more then 2 VLANs via UCI for mesh-connections and add them to bat0 the network stack isn't working correctly any longer. Any invocation of ip, ifconfig, brctl, batctl etc will result in a stuck system.
If I configure just two mesh-VLANs the system is working fine. It is also possible to add more VLANs to bat0 via "vconfig eth0.x" and "batctl if add eth0.x".

The strange part is before the crash happens the kernel always prints "tried to remove device eth0 from br-client" but eth0 itself shouldn't be part of any bridge or whatever. I also don't see this message when I configure just two mesh-VLANs.

See attached files for backtrace and /etc/config/network.
If you need further information please let me know.

[dmesg.txt](https://github.com/freifunk-gluon/gluon/files/165051/dmesg.txt)
[network_config.txt](https://github.com/freifunk-gluon/gluon/files/165050/network_config.txt)

NeoRaider, 2016-03-09:
Interesting, I'll try to reproduce this with other hardware.

Which batman-adv version are you using, compat 14 or 15?

belzebub40k, 2016-03-09:
root@0216088335f8:/# batctl -v
batctl 2016.0 [batman-adv: 2016.0]

As far as I understand this refers to compat 15.

belzebub40k, 2016-03-10:
It looks like this problem is some how related to br-client. If I remove br-client completely from /etc/config/network I'm able to configure 4 mesh-VLANs (didn't try more).

However this isn't really an option as this device is intended to be a backbone node with connects several Ubiquiti devices running stock firmware. We have a management VLAN (3) on each of them which is then bridged with bat0 via br-client. Without this we wouldn't be able to access them from remote.

NeoRaider, 2016-03-10:
I've forwarded the issue to the batman-adv developers, they'll have a look.

belzebub40k, 2016-03-10:
Thanks. Is there a ticket I can track and contribute to?

I just found some more details that may be helpful:
I can leave br-client enabled but need to remove bat0 from it. And again it wont harm the system if I add it manually once everything is up.

NeoRaider, 2016-03-11:
I think there is no ticket yet, I'll open one.

I wasn't able to reproduce this issue on x86-vmware with the current Gluon master.

It might be worth noting that the crash happens when the VLAN interfaces are *removed* from bat0. netifd first sees eth0 as up, adding the 4 VLAN interfaces to bat0, one second later it is seen as down again; when eth0.102 is removed from bat0, something fails and the kernel instead tries to remove eth0 from br-client, which causes the crash as eth0 is not part of br-client.

belzebub40k, 2016-03-13:
Can confirm that. The problem happens when the interface eth0 is going down. When I set everything up manually and then power cycle the switch it crashes with the same error message.

NeoRaider, 2016-03-18:
https://www.open-mesh.org/issues/247

belzebub40k, 2016-03-18:
Could this be the same problem as we have here?
http://patchwork.ozlabs.org/patch/587118/

belzebub40k, 2016-03-18:
Ok tried the attached patch against master and it no longer crashes because of trying to remove eth0 from br-client. Now if fails because it tries to remove eth0 from local-node. Looks like I'm digging in the right direction.

[dmesg.txt](https://github.com/freifunk-gluon/gluon/files/180240/dmesg.txt)
[patch.txt](https://github.com/freifunk-gluon/gluon/files/180241/patch.txt)

belzebub40k, 2016-03-18:
Ok, after some more testing I recognized current 2016.1.2 images without next_node are working fine. It looks like its a combination of to many tagged vlans and macvlans that causes the issue.

NeoRaider, 2016-03-25:
@belzebub40k, can you test the updated patch from http://lists.openwall.net/netdev/2016/03/25/103 ?

NeoRaider, 2016-03-25:
Ah, it can also be found in patchwork: http://patchwork.ozlabs.org/patch/602124/

belzebub40k, 2016-03-25:
Will try tomorrow.

belzebub40k, 2016-03-28:
Perfect until now it seems to work without problems. I will give it some more time for testing and keep you posted.

belzebub40k, 2016-04-06:
Sadly it does work properly. It crashes with the same error as reported initially when the counterpart is power cycled. However disconnecting it physically doesn't cause a crash.

Today I got my BananaPro and I tried unpatched (except those to profiles.mk to build the image) v2016.1.3 on it and it doesn't crash which is very strange as it is mostly a BananaPi M1 with a WiFi chip.

I'm at a point where I consider faulty hardware. I try to get my hands on an other hardware revision of the M1. The one I have is a clone from sinovoip and not the original from Lemaker.

NeoRaider, 2016-04-06:
I think the issue is that the kernel code is completely b0rked and may fail in various spectacular ways.

NeoRaider, 2016-10-20:
As there is an upstream fix now, I'll try backporting it to the Gluon kernels.

poldy79, 2016-10-20:
As we might see this on our gateways, do you have a link to the upstream ticket?

T-X, 2016-10-20:
Two upstream fixes related to interface removal:

- [batman-adv: don't access unregistered net_device object](https://git.open-mesh.org/batman-adv.git/commit/958aafba93c7e408237298c5b2c5d7c3e318402c) (present since v2015.2)
- [batman-adv: Modify mesh_iface outside sysfs context](https://git.open-mesh.org/batman-adv.git/commit/a1f0a804cacda8ca0847cbac07adcbf95e9c24c1) (master branch only, no release with it yet, not queued for stable kernels?)
  - History:
    - [batman-adv: Fix hardif remove/add race](https://git.open-mesh.org/batman-adv.git/commit/6b0485c758be31ff3d466644975eaedc54db6d17)
    - [batman-adv: postpone sysfs removal when unregistering](https://git.open-mesh.org/batman-adv.git/commit/a33c882c10692b99ce647c929cb90c62e87fa083)
   - Then both reverted:
     - [Revert "batman-adv: Fix hardif remove/add race"](https://git.open-mesh.org/batman-adv.git/commit/b58db7bfdadf59c4a148591fce7104e236f2c8ce)
     - [batman-adv: Revert "postpone sysfs removal when unregistering"](https://git.open-mesh.org/batman-adv.git/commit/0994697e926d9633adb4912036548b0bb1d110ea)

Still pending is this ticket with an according patchset to fix this which still needs a review:
- https://www.open-mesh.org/issues/168
- https://patchwork.open-mesh.org/patch/16718/
- https://patchwork.open-mesh.org/patch/16719/

(no claim that this list is complete - @neoraider, if you noticed any other patches feel free to add :-) )

belzebub40k, 2016-10-26:
@NeoRaider I did a dirty backport of Andrew Collins kernel patch and it looks much better. However now it seems like there is a race condition caused by eth0 getting up and down quickly. This sounds exactly like the problem addressed with the batman-adv patches mentioned by @T-X . I'll try with batman-adv master instead of 2016.2.

belzebub40k, 2016-10-27:
Did not figure out how to change to batman-adv master thus just applied https://git.open-mesh.org/batman-adv.git/commit/a1f0a804cacda8ca0847cbac07adcbf95e9c24c1 as a patch. That seams to fix the race condition mostly. At least it does not crash the kernel when uplugging and re-plugging eth0. But I still see messages which indicate that the race condition isn't fixed fully.

batman_adv: (null): no_rebroadcast: Changing from: disabled to: enabled
NOHZ: local_softirq_pending 08

The "batman_adv (null)" messages appear during every restart of eth0. "local_softirq_pending" happend just one time and I wasn't able to reproduce it a second time by now.

**console log boot and restart of the interface triggered by unplugging or rebooting the switch**
[console.txt](https://github.com/freifunk-gluon/gluon/files/556627/console.txt)

**applied patches**
[batman-adv-modify-mesh_iface-outside-sysfs-context.txt](https://github.com/freifunk-gluon/gluon/files/556638/batman-adv-modify-mesh_iface-outside-sysfs-context.txt)
[netdev-all_adj_list-refcnt-propagation-to-fix-panic.txt](https://github.com/freifunk-gluon/gluon/files/556639/netdev-all_adj_list-refcnt-propagation-to-fix-panic.txt)

NeoRaider, 2017-02-23:
Both mentioned patches are included in Gluon now (and I've just backported the batadv patch to -legacy), so I guess we can consider this fixed.

closed 2017-02-23 by NeoRaider

Adorfer, 2017-05-28:
BTW: Ganz blöd, wenn das während eines Autoupdates auftritt.
https://paste.debian.net/947029/