freifunk-gluon/gluon #683: add wifi config mode

mokkin, 2016-03-11:
Some devices without a LAN port (such as TL-WA854RE) cannot be flashed easily, because the config mode is not accessible without cable.
The config mode should open a config wifi to access the device.

NeoRaider, 2016-03-11:
The security concept for the config mode is based on physical access. There is no way to establish security on devices without ethernet in a sane way. Of course, we could add hardware-specific hacks for such devices (for example using the same default password the stock firmware uses), but we usually don't think such hacks are worth the extra maintenance work.

A config WLAN won't happen in Gluon. There is absolutely no value for us in supporting such devices, as there are tons of similar devices with ethernet which aren't significantly more expensive.

While we won't officially support such devices, they could be added as BROKEN; users would need to compile Gluon themselves with the following settings:
* Set `setup_mode.skip = true` in site.conf
* Add gluon-authorized-keys and configure a SSH key

mokkin, 2016-03-11:
I understand it's not worth the extra work for such few devices.

But thanks for the hint to skip the conifg mode in site.conf. This makes it relatively easy to create a custom image for this job.

FreifunkUFO, 2016-04-01:
it would be a nice value to configure mesh nodes via wifi!
You fear about security? so the "WAN" could be deactivated within config-mode
broken device? no, its the almost the same shit as tplink 841, but without ethernet

A-Kasper, 2016-04-01:
it would be nice to launch config mode via ssh and access it via wlan at
nextnode address for all devices. there are many hard to reach mesh only
devices wich would be much easier to maintain if config mode would be
available without phyical access. if someone really want to restrict to
physical access, there should be an option in site.cfg to do so.

kb-light, 2016-04-03:
@A-Kasper why not directly using ssh for reconfiguring a hard to reach mesh only node?

A-Kasper, 2016-04-03:
usability. but you're right: it should be possible to launch configmode via
webbrowser+password

viisauksena, 2016-04-18:
scary ideas ... as the wifi avaibility is the same as in the whole freifunk net of your comunity ... 
you really dont want to have this ... 
for basic config you could compile with no_config_mode in site.conf - so you get a basic config which could work, and do the config afterwards via link local or via your freifunk network and given ssh-keys which should be implemented - (and could be deleted if wished after config) - its a bit of a hack, but not as impossible to build and config

NeoRaider, 2016-07-27:
As we can't support HTTPS (there's no space for a TLS library on 4MB devices), adding password authentication to the setup interface is not an option. I don't think we want to support devices without LAN interface anyways, so I'm closing this.

We have some ideas to add a secure online configuration interface (see http://www.nilsschneider.net/2015/08/14/ff-online-config.html ) though.

closed 2016-07-27 by NeoRaider

schlady, 2016-07-27:
@NeoRaider has there been any effort on implementing the secure online configuration interface yet?

viisauksena, 2016-07-27:
@schlady i heavily recommend ssh for this, this is secure and meant for that kind of things, and works like a charme, even if terminal stuff is scary for some people