freifunk-gluon/gluon #686: Turn off dropbear if no access is configured

mmalte, 2016-03-13:
Dropbear is always running, even if there no ssh key or password configured.

Turning it off would save RAM and improve security.

It would also allow us to easily, even remotely, make sure that nobody has access to a node.

A-Kasper, 2016-03-14:
it would be a nice idea to put a ssh switch on the statuspage to launch
dropbear only if needed.

jplitza, 2016-03-14:
I like @mmalte's initial proposal, but don't understand the need for what @A-Kasper proposed. Is it to save memory even on nodes with remote access configured? Because security-wise, it obviously has no benefit, and usage-wise it may even be problematic because there are some connections so shitty that SSH may barely work, but the status page doesn't load.

A-Kasper, 2016-03-14:
it's just an idea to free some more memory

RubenKelevra, 2016-09-09:
Actually, a button which is capable of enabling a service via web interface is more likely to be implemented with a security flaw than dropbear is contain a remote exploit which requires no login.

J0WI, 2016-09-09:
dropbear should only run, if a password or public key is defined.
If no password but at least one public key are defined, it should only allow public key authentication.

MPW1412, 2017-12-30:
I put some thoughts into this. I see two possible ways to achieve this:

a) Disable and enable password login or even the whole daemon in gluon-web-admin/..../remote.lua depending on the user's input in config-mode. The uci entries for this are
```
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
```
and they are already properly parsed by dropbear's init script. This could be set by lib/gluon/web/model/admin/remote.lua when parsing user's input.

There is no uci field which determines, weather a key is set. So the whole daemon needs to be disabled if neither a password nor a key is set.

All this logic could happen in remote.lua.

b) Parsing both /etc/shadow to check for a password and /etc/dropbear/authorized_keys in dropbear's init script, apply the needed configuration and start the daemon if required.

Solution a) would result in a more complex command line administration. You'd not only have to set the password, but also change the uci entries in case of ssh changes. Respectively enable or disable the daemon manually, if a key is added or removed.

For those reasons, I dislike solution a). Solution b) could be implemented in Gluon, but needs a fork of the lede package "dropbear" to change the init script. This leads to an unnecessary complication in case of security updates for dropbear.

My conclusion of these thoughts is, that this issue should be closed here and reopened in LEDE directly.

Regards,
Matthias

CodeFetch, 2019-06-27:
The only solution that makes sense to me is to disable dropbear by default and add a checkbox to the config mode to enable it and only show the public key/password fields when it is enabled.

Letting the init script or the config mode page check whether a password or public key is set is unnecessary.

PR #3541 (2025-07-12): gluon-harden-dropbear: Add package
> Upon gluon-reconfigure, this package disables password-based SSH access, if the account root is locked. It furthermore disables the dropbear service if neither passwords nor authorized keys are present.

This intends to resolve #686.

I think `gluon-reconfigure` is a decent compromise between not having to fiddle with the dropbear settings manually and us definitely not wanting to patch the init script of dropbear.

I intend to take a look at also closing the firewall port if dropbear is offline.
closed 2025-07-12 by neocturne