freifunk-gluon/gluon #687: Picostation startet immer wieder mit leeren Einstellungen

FabianCernota, 2016-03-14:
Hallo zusammen,
ich habe auf eine Picostation Gluon v2016.1.2 geflasht. Wenn ich jetzt im Configmode bin und Einstellungen setze sind diese nach einem Neustart wieder weg und ich lande im Configmode.

Muss ich noch irgendetwas machen?

Gruß
Fabian

NeoRaider, 2016-03-14:
Dein Gluon-Image ist kaputt; es wurde beim Update von v2016.1 auf v2016.1.2 entweder `make update` oder `make clean` (oder beides) vergessen.

In diesem Zustand ist der Flash nicht beschreibbar, also auch kein normales Sysupgrade möglich. Du musst nach Erzeugen eines korrekten Images dieses per TFTP flashen.

FabianCernota, 2016-03-24:
Auch mit einem frischen Gluon kann ich es noch auf 2 verschiedenen Picostations nachstellen. 

Im Forum gibts es noch jemand der auch Probleme mit der Picostation hat:
https://forum.freifunk.net/t/ubiquiti-picostation-m2-startet-nur-in-den-gluon-configmode/11361

NeoRaider, 2016-03-24:
@FabianCernota, wie hast du das frische Gluon aufgespielt? Wenn schon einmal ein kaputtes Gluon drauf war, muss über TFTP geflasht werden.

FabianCernota, 2016-03-24:
@NeoRaider  Eine per TFTP und eine von der Stockfirmware heraus.

NeoRaider, 2016-03-24:
@FabianCernota Logs?

FabianCernota, 2016-03-24:
@NeoRaider Welche genau? An die Gluon Logs komme ich ja nicht.

NeoRaider, 2016-03-24:
@FabianCernota, warum nicht? Einfach per Telnet drauf und `dmesg` bzw. `logread` eingeben.

FabianCernota, 2016-03-26:
http://pastebin.com/ZUrFTJ5v

NeoRaider, 2016-03-26:
Okay, die Picostation benutzt einen Flash-Chip von einem anderen Hersteller als Macronix... ich bau nachher nen Patch.

MPW1412, 2016-03-26:
Wir haben das hier auch mehrfach schon gesehen jetzt, falls du noch Logs brauchst. Wir freuen uns auf den Patch :).

NeoRaider, 2016-03-26:
Im master sollte 1ccd24df770f24d5660885d264b3a2af51bf6d1e das Problem beheben. Bitte einmal testen, wenn der Patch funktioniert, landet er auch in v2016.1.x.

NeoRaider, 2016-03-26:
Ich habe gerade einen anderen Patch und zugehörige Diskussionen gefunden, daher vermute ich, dass mein Patch mehr Probleme auslöst als er behebt.

Siehe:
http://patchwork.ozlabs.org/patch/549173/
http://patchwork.ozlabs.org/patch/553683/

NeoRaider, 2016-03-26:
Könnte mir jemand leihweise eine entsprechende PicoStation zukommen lassen? Adresse gibt's per Mail oder IRC.

NeoRaider, 2016-03-26:
Hmm, CPE210 und WDR4300 benutzen denselben Flash-Chip. Ich probier erstmal, das Problem mit den Geräten zu reproduzieren.

NeoRaider, 2016-03-26:
Bitte doch einfach mal mit dem Patch testen (also nicht den neuesten Stand des Master, in dem ich den Patch reverted habe, sondern den davor).

NeoRaider, 2016-03-26:
Und könnte mal jemand eine PicoStation aufschrauben und nachschauen, was auf dem Flash-Chip steht? Laut Kernel-Log ist das ein Spansion s25fl064k, aber möglicherweise ist der Log flasch und es ist eigentlich ein Winbond w25q64.

justelex, 2016-03-27:
Hi,
ich habe die eben mal aufgeschraubt, 
es ist ein Winbond 25Q64FVIG
Danke für deine Mühe

NeoRaider, 2016-03-29:
Neuer Patch: d9ca449ed0109651d748b22e6a6421867b1e9ae9 Bitte testen!

FabianCernota, 2016-03-29:
@NeoRaider Funktioniert leider noch nicht:

`[    0.000000] Linux version 3.18.27 (cernota@arch-fabian) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r47335) ) #1 Tue Mar 29 22:46:54 CEST 2016
[    0.000000] MyLoader: sysp=cc33cc33, boardp=33cc33cc, parts=cc33cc73
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x01ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x01ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x01ffffff]
[    0.000000] On node 0 totalpages: 8192
[    0.000000] free_area_init_node: node 0, pgdat 803742b0, node_mem_map 81000000
[    0.000000]   Normal zone: 64 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line:  board=UBNT-BM console=ttyS0,115200 mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 28324K/32768K available (2573K kernel code, 133K rwdata, 540K rodata, 264K init, 193K bss, 4444K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:390.000MHz, DDR:390.000MHz, AHB:195.000MHz, Ref:5.000MHz
[    0.000000] Calibrating delay loop... 259.27 BogoMIPS (lpj=1296384)
[    0.070000] pid_max: default: 32768 minimum: 301
[    0.070000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.090000] NET: Registered protocol family 16
[    0.090000] MIPS: machine is Ubiquiti Bullet M
[    0.100000] registering PCI controller with io_map_base unset
[    0.320000] PCI host bridge to bus 0000:00
[    0.330000] pci_bus 0000:00: root bus resource [mem 0x10000000-0x13ffffff]
[    0.330000] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.340000] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.340000] pci 0000:00:00.0: [168c:002a] type 00 class 0x028000
[    0.340000] pci 0000:00:00.0: fixup device configuration
[    0.350000] pci 0000:00:00.0: reg 0x10: [mem 0x10000000-0x1000ffff 64bit]
[    0.350000] pci 0000:00:00.0: supports D1
[    0.350000] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    0.350000] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.350000] pci 0000:00:00.0: BAR 0: assigned [mem 0x10000000-0x1000ffff 64bit]
[    0.350000] pci 0000:00:00.0: using irq 40 for pin 1
[    0.360000] Switched to clocksource MIPS
[    0.360000] NET: Registered protocol family 2
[    0.370000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.370000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.380000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.390000] TCP: reno registered
[    0.390000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.400000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.400000] NET: Registered protocol family 1
[    0.410000] PCI: CLS 0 bytes, default 32
[    0.410000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.440000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.450000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.460000] msgmni has been set to 55
[    0.460000] io scheduler noop registered
[... 114 lines cut ...]
[   28.910000] jffs2: Newly-erased block contained word 0x1ad33f51 at offset 0x000d0000
[   28.920000] jffs2: Newly-erased block contained word 0x2fe9519e at offset 0x000c0000
[   28.930000] jffs2: Newly-erased block contained word 0x88142a65 at offset 0x000b0000
[   28.960000] jffs2: Newly-erased block contained word 0xccbe88ad at offset 0x000a0000
[   28.970000] jffs2: Newly-erased block contained word 0x59b24a94 at offset 0x00090000
[   28.980000] jffs2: Newly-erased block contained word 0x5ceff193 at offset 0x00080000
[   28.990000] jffs2: Newly-erased block contained word 0x47222d48 at offset 0x00070000
[   29.000000] jffs2: Newly-erased block contained word 0xdeb83c1d at offset 0x00060000
[   29.030000] jffs2: Newly-erased block contained word 0x23020c34 at offset 0x00050000
[   29.040000] jffs2: Newly-erased block contained word 0x2023d8fa at offset 0x00040000
[   29.050000] jffs2: Newly-erased block contained word 0x9ca5caaa at offset 0x00030000
[... cut ...]
[   29.110000] jffs2: notice: (1072) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   33.030000] device eth0 entered promiscuous mode
[   33.030000] IPv6: ADDRCONF(NETDEV_UP): br-setup: link is not ready
[   34.130000] eth0: link up (100Mbps/Full duplex)
[   34.130000] br-setup: port 1(eth0) entered forwarding state
[   34.140000] br-setup: port 1(eth0) entered forwarding state
[   34.160000] IPv6: ADDRCONF(NETDEV_CHANGE): br-setup: link becomes ready
[   36.140000] br-setup: port 1(eth0) entered forwarding state
[   68.290000] random: nonblocking pool is initialized
`

NeoRaider, 2016-03-29:
@FabianCernota, den Kernel-Tree hast du aber gecleaned nach dem git-pull/make-update, oder?

FabianCernota, 2016-03-29:
Jap git pull , make clean, make update

NeoRaider, 2016-03-29:
Hmm, seltsam. Dann komme ich ohne Testgerät nicht weiter.

FabianCernota, 2016-03-29:
Ich lass dir eine zukommen alles weitere per Mail dann :)

MPW1412, 2016-03-31:
Scheinbar kann man das Problem umgehen, wenn man erst AirOS auf eine alte Version verringert. 

Querverlinkung: https://forum.freifunk-muensterland.de/t/picostationm2-v2016-1-1-0-0/884/8?u=mpw

NeoRaider, 2016-03-31:
@MPW1412, ja, das ist allgemein bekannt. Es geht in diesem Report darum, diesen Workaround in Zukunft vermeiden zu können.

MPW1412, 2016-03-31:
Also ist das primäre Problem gar nicht der neue Flash-Chip, sondern irgendwas im Bootloader, was den Flash nicht korrekt entsperrt.

NeoRaider, 2016-03-31:
Es ist eigentlich nicht die Aufgabe des Bootloaders... der Kernel sollte das schon selbst können.

SilversurferCH, 2016-04-06:
Wenn du noch beim Testen Hilfe brauchst, hab hier auch noch eine betroffen Pico liegen.

LeSpocky, 2016-04-06:
Bestätige Winbond 25Q64FVIG auf PicoStation M2 XM.

A-Kasper, 2016-04-06:
Passiert der Fehler auch wenn ihr vorher nen downgrade gemacht habt?
Ich hatte das auch erst. Habe dann per tftp zurück auf stock und von da
auf 5.5 und dann auf gluon. dann ging es.

NeoRaider, 2016-04-06:
@A-Kasper, nein, der alte Bootloader entsperrt den Flash. Das Problem tritt nur mit dem neuen auf.

txt-file, 2016-04-18:
Am 11. April kam eine neue Version von 5.6.x raus. Das Changelog klingt so, als wäre dort dieser Bug behoben.
Hat jemand Zeit und Gerät um das einmal zu testen?

https://www.ubnt.com/downloads/firmwares/XN-fw/v5.6.4/changelog.txt

NeoRaider, 2016-04-19:
@txt-file, das wäre wieder nur ein Workaround, der echte Fix muss im Kernel passieren. @FabianCernota, wie sieht's mit dem Testgerät aus?

SilversurferCH, 2016-04-20:
Ich habe gerade mal mit der 5.6.4 getestet. Leider gleiches Ergebnis. Gluon war 2016.1.3.

closed 2016-05-06 by NeoRaider

LeSpocky, 2016-05-28:
Erfolgreich getestet mit gluon v2016.1.5: https://map.md.freifunk.net/#!v:m;n:68725142b11b :beers:

viisauksena, 2016-09-12:
ich habe mit unserem letzten update alle betroffenen cpe210 v1 und v1.1 abgeschossen .. symptom : Geräte tot, leuchtet nur noch power - meist nur config mode
der upgrade passierte von 
v2016.1.5.u / gluon-v2016.1-183-gfb8c7a8
auf
v2016.1.6 / gluon-v2016.1-295-gc673356
mit den 841 egal welcher art gibt es keine Probleme mit der FW ... ich wüsst nicht wie ich da noch mehr Info liefern könnt.. ich Versuch nochmal ganz neue FW zu bauen und die zu flashen

NeoRaider, 2016-09-12:
@viisauksena 

1. Was hat das mit diesem Ticket zu tun??? #796 war das einzige CPE-spezifische Ticket in letzter Zeit... und wenn du dir unsicher bist, mach lieber ein neues Ticket auf.
2. Was sind das für Versionsnummern? v2016.1.6 kommt aus dem Stable-Branch, v2016.1-295 muss ein Master-Commit sein (im Stable-Branch gab es seit v2016.1 nur 83 Commits, keine 295)
3. Möglicherweise kein `make clean` nach dem `make update` ausgeführt? Das würde das Problem in diesem Fall erklären.

viisauksena, 2016-09-12:
das sind die bezeichnungen aus dem meshviewer, der erste Teil ist selbstgewählt und der letzte kommt aus dem Masterbranch , korreckt - da die beiden Plattformen ähnlich (so dachte ich - und das oben auch erwähnt wird)  sind hab ich das mal so aufgemacht. 
ich dachte ich probier noch ein paar Dinge bevor ich Alarm schlage

PolynomialDivision, 2020-09-03:
Habt ihr zufällig ähnliche Probleme mit nem 5.4 Kernel auf AR9342? Man findet diesen issue über: https://dev.archive.openwrt.org/ticket/20982

commit 40d087de (2016-05-06): kernel: backport spi-nor driver from 4.4.9

Fixes #687
commit 1f400189 (2016-05-09): kernel: backport spi-nor driver from 4.4.9

Fixes #687