freifunk-gluon/gluon #689: request: UbiFi AP AC

Patschke, 2016-03-15:
My student hostel is using UbiFi AP AC access points by Ubiquiti (Model: UAP-ACv2). If a Gluon firmware is created for them, we could join the Freifunk network. We got full access to the APs, but are not allowed to exchange them. 

Support already exists for some other Ubiquiti devices (e.g. UniFi AP, UniFi AP Outdoor), but does not for that specific one. Since the AP AC is quite expensive i don't expect a huge user community. Mabe it's still worth to add support for it, especially if it's similiar to the already supported ones.

rotanid, 2016-03-15:
although i can't help with implementation, there are two things to mention:
1) some more information might be necessary (a more exact definition of the Hardware you are talking about) as there are older access points and newer ones with the label "AP AC" from Ubiquiti and at least the newer ones are available in four different "flavours"
2) if they aren't supported by OpenWRT, it's rather unlikely gluon, which is based on OpenWRT, can do anything to change that

Patschke, 2016-03-15:
1) the label at the back says: UniFi AP AC, M/N: UAP-AC, FCC ID: SWX-UAPAC, IC: 6545A-UAPAC. The info command returns: Model: UAP-ACv2, Version 3.3.19.4015 (i guessed the last one is the firmware-version?)
2) Going to check that

NeoRaider, 2016-03-18:
The UAP AC Lite is supported by the OpenWrt trunk since http://git.openwrt.org/?p=openwrt.git;a=commit;h=8176e7d23b8da89febabf55a8d2576973658c248 ; if the LR/PRO versions are identical regarding firmware is unclear.

Anyways, these devices use ath10k-based 5GHz WLAN, which doesn't work well for mesh networks at the moment (this might be different when 802.11s is used instead of IBSS mode, but we haven't tested it yet.)

rotanid, 2016-03-18:
@NeoRaider googling the FCC ID results in informations about the old "UAP AC" from 2013, not the newer ones like Lite/LR/PRO that you mention

NeoRaider, 2016-03-18:
The old version is based on Broadcom, AFAIK completely unsuitable for mesh networks. If I remember correctly, that versions also used a signed firmware, so no custom firmware can be flashed...

Patschke, 2016-03-19:
They allow custom firmware, but if they are unsuitable for mesh networks i guess that doesn't matter. 

I hoped they are close to already supported ones, but since this seems not to be the case, i'll close this request. Thanks for your replies, and thanks for your great work on Gluon!

plumpudding, 2016-05-29:
@NeoRaider I would like to reopen this for the Lite. I tried the broken image and it doesn't work at all: The LAN interface seems to be down (no multicast responses) and the LEDs don't light up at all.
I noticed that only the Pro is referenced [here](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/files/lib/gluon/upgrade/020-interfaces) and [here](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/files/lib/gluon/upgrade/020-interfaces)
I tried adding the Lite but without any success.

NeoRaider, 2016-05-29:
@plumpudding, the files you referenced are correct as the are: They switch the LAN ports of the Pro, but he Lite has only one.

Does the failsafe mode work on the Lite?

plumpudding, 2016-05-29:
No, it doesn't. But I had trouble getting into failsafe mode with LEDE too.

commit e773f686 (2016-07-27): ar71xx-generic: fix BOARDNAME for UAP AC Lite

Caused by a broken backport.

Fixes #689
closed 2016-07-27 by NeoRaider