freifunk-gluon/gluon #691: backport TP-LINK WR710N v2.1

cbricart, 2016-03-16:
TP-LINK WR710N v2.1 (EU) has been spotted in the wild...
actually, this hardware is a TL-WR710 v2 with 8MB rather than 4MB Flash which is supported in DD-Trunk since https://dev.openwrt.org/changeset/47849 and a backport to CC (according to  https://forum.openwrt.org/viewtopic.php?id=59290 ) seems to be just a patch to the `ar71xx_generic` Makefile ..

is there any chance to integrate this hardware to the current 2016.1.x relese cycle?

NeoRaider, 2016-03-17:
I don't really like that the upstream patch doesn't update `target/linux/ar71xx/base-files/lib/ar71xx.sh` to correctly set the model string to contain "v2.1" even though the devices use different images; in particular, this will cause the autoupdater to use the "v2" image. This would work because OpenWrt's image compatiblity check won't notice and the images are compatible, but it is not really nice: if we later fix the image compatiblity check to be more thorough, this will lead to v2.1 devices that will only accept v2 images after the v2 image has been flashed once...

Backporting this is not a problem, but I'll need to think about this a few days to decide if I'd like to fix this upstream, or just not care.

NeoRaider, 2016-03-25:
I've sent a few patches for the model detection upstream, I'll backport the support as soon as I get feedback.

FreifunkUFO, 2016-04-01:
where are your patches? please link the gluon ticket vs. openwrt ticket!

NeoRaider, 2016-04-01:
@FreifunkUFO, the patches have been applied upstream in r49105-r49107.

NeoRaider, 2016-04-01:
Fixed in 3732c26e95b758503484beced62e01d1273d25d2, please test.

closed 2016-04-01 by NeoRaider

FreifunkUFO, 2016-08-05:
very bad idea to call it "v2.1"! 
That dot "." is evil, better use "-"!

NeoRaider, 2016-08-08:
@FreifunkUFO: Nah, we generally don't escape ".", as many image revision identifiers use . and we want the image names to have "natural" names.

FreifunkUFO, 2016-08-08:
ok :-/ 
who is "we"?

NeoRaider, 2016-08-08:
Basically, the libplatforminfo code used by the autoupdater (which was developed by @tcatm and me). While it would be possible to define the image names independently of the model indentifier strings (as the manifest file defines the mapping of these), I don't see any reason to change our current naming.

What is the reason you don't like the "."? Regex issues?

FreifunkUFO, 2016-08-10:
the "." is "evil" and i will ask openwrt/lede for changing the filename.
(libplatforminfo & autoupdater are only gluon-specific things..)

NeoRaider, 2016-08-10:
The "." is part of the revision name and a standard ASCII character, so the filename should match that. It doesn't have any special meaning in shell (I'd agree that ", ', *, ?, etc. might be problematic) or URLs, so I don't understand at all what you are talking about.

Note that Gluon image naming doesn't follow the image naming in OpenWrt/LEDE.

FreifunkUFO, 2016-08-31:
hm, so i have to ask openwrt/lede

NeoRaider, 2016-08-31:
1. As written, we have our own naming; changing the names in OpenWrt/LEDE won't have an effect on Gluon (except that we'd need to update our profile definitions)
2. I've asked you several times what issue you see with the "." character, which you only answered by calling it "evil".
3. Your request has nothing to do with this ticket (and we already have several image names that contain ".")