freifunk-gluon/gluon #694: sbin/autoupdater

viisauksena, 2016-03-19:
where do i find /sbin/autoupdater 
i would love to add a -F flag (uppercase) which do a autoupdate even if check "think" it has latest,
i would do it by emptying release file 
```echo > /lib/gluon/release```

reason : i often have realy minor updates in FW and than i wouldnt do a extra release for that, 
reason2: if you mistakenly make a FW with v2106.1.2 , you have to be tricky 

with that patch you could use existing autoupdater structures and not invent the wheel new, or do a  ```wget FW -O /tmp/somefile ; sysupgrade /tmp/somefile``` by hand , or at least for every different hardware model

kb-light, 2016-03-19:
I think you are searching this one: [https://github.com/freifunk-gluon/packages/tree/master/admin/autoupdater](url)

To me '-F' doesn't make sense, because it will just work for manually updates and not for updates executed by gluon-autoupdater. And in case you are just doing manually updates, you can easily manipulate the version number within the manifest, before signing it.

NeoRaider, 2016-03-19:
I'd call such a flag `--force-downgrade`, and it should not work by moving files away, but just change the version check. I don't think implementing this in the Lua autoupdater makes much sense though, as we've started implementing a new version in C (not online yet.)

As this issue belongs in the "packages" repo and Github doesn't allow to move it, I'm closing this.

closed 2016-03-19 by NeoRaider

viisauksena, 2016-03-19:
see : https://github.com/viisauksena/packages-1/commit/3d50f722856fa95025e8f64a6ec7b8c6b9d65198