freifunk-gluon/gluon #695: 2016.1.x autoupdate : 20% did not turn up again

viisauksena, 2016-03-21:
so, i am hunting the reasons now .. (and edit this : when i get more info in these days)

while nodes get there autoupdate normaly, some of them did not come back. The Nodes are 841v9 and 841v10 .. there are updating from a 2-3 week old master to a 5 day old master ... 
Source is generaly here: https://cccfr.de/git/viisauksena/firmware/tree/support/

one Node at my home give the hint : it was without reason in config mode - this could be by accident, i will go to the places where the other nodes are installed and try to find out more .. 

maybe its related to this, also not so precise bug description: https://github.com/freifunk-gluon/gluon/issues/635
http://oi66.tinypic.com/4pxxrq.jpg
http://oi68.tinypic.com/5xu6tk.jpg
http://oi64.tinypic.com/29kthqa.jpg

NeoRaider, 2016-03-21:
Most likely #669.

viisauksena, 2016-03-21:
that issue relate to firmware build - i thought that too big firmware isnt simply build at all ? 
therefore you have to add the newer v9 and v10 of tplink841 in the list of affected ... 
this is strange because on many other routers (also v9 and v10 it does not happen) 
the also mentioned (in #669 ) openwrt bug is power related .. which seems also very unlikely - while the routers are fixed positioned on places where no one could touch anything

i checkout in which state the affected routers are and try to get dmesg and so .. 
anythink else you suggest me to look for?

NeoRaider, 2016-03-21:
The issue occurs completely at random, not correlated with the size of the firmware. We don't know of any conditions that make the issue more or less likely. So far, we've seen it on different revisions of TL-WR841 (and the OpenWrt tickets also mention a few other devices), but it seems likely that any device could be affected.

viisauksena, 2016-03-21:
i am totaly unhappy / unsatisfied with a issue happening randomly on deterministic mashines :-(

NeoRaider, 2016-03-21:
And so are we... in theory, these MIPS machines shouldn't have any entropy to behave nondeterministically in early boot. This makes quirky hardware (triggered by faulty software) the most likely explanation.

As timing affects the issue, it is possible that a delay is missing when the serial console in reconfigured. Another possible explanation is that the serial controller doesn't like that there are two consoles (ttyS0 and early0) registered at the same time, or that ttyS0 is unregistered without being registered before. I'll need to review the code further and ask the respective maintainers about their opinions on this issue.

viisauksena, 2016-03-22:
some more information about the lost nodes ...
3 times were totaly dead : no bootup - no connection , no configboot (gluon), no failsafe/rescue-boot (openwrt) ... tftp helped : this might be your ( @NeoRaider ) described error.
2 times , no freifunk mesh and so on, boots up but no bridge if , no batman if ... but br-setup - looks a bit like it killed itself in the middle of a sysupgrade - firstboot and reboot helps (but then with new fastd keys)

so both not necessarily the same root - but very annoying outcome of autoupdater ... i guess this fails in 1 out of 10 processes. (if i take the last 5 autupdate processes of the 35 heavy monitored router)

mmalte, 2016-03-22:
We are running stable autoupdate to 2016.1.2 since a few day, since yesterday for pre 2016.x nodes.

About 400 stable nodes so far, before that ~100 beta/exp nodes. I suspect that the update failed on one 841v7, one 941v6 and one 841v9 because they didn't show up again after downloading the image.

This is not to unusual, as long as they'll come back after a reboot. Unfortunately the nodes owners did not respond to my mails yet.

viisauksena, 2016-03-22:
@mmalte the interesting part will be - in which state are these routers if they dont come back : and if you can, can you safe the last ram:usage stuff from nodes.json (yes , there should this info still laying around, even if node is offline and it is not shown in meshviewer) . just to sort out some ram-usage problems of failing sysupgrade. (which seems slightly different than the magic-bug reported by @NeoRaider )

kpanic23, 2016-03-22:
Well, in our community (ff3l) we had not one bricked router on the last update. On the update before that it was only one bricked router (which supposedly had some kind of interruption during the update). At the moment we have roughly 600 nodes.
@viisauksena Do you use "unset SED" before building the firmware? Since unsetting SED we had'nt had any problems any more.

viisauksena, 2016-03-22:
was meinst du mit **unset SED**

kpanic23, 2016-03-22:
According to https://wiki.openwrt.org/doc/howto/build :

environment variables:
SED should not be set. If it is, run `unset SED` before compiling.
GREP_OPTIONS should not have –initial-tab or other options affecting its output

Actually, I don't know if this is still needed, but I guess it doesn't hurt...

NeoRaider, 2016-03-24:
Closing this as duplicate of #669. Please comment if you find evidence that this is an unrelated issue.

closed 2016-03-24 by NeoRaider

viisauksena, 2016-03-25:
i am not 100% conviced that this is a pure update race-condition error.
because its unexplained why these routers fail everytime to come back after sysupgrade - and do not answer anything.(no openwrt failsafe, no gluon configmode).
but when they are reflashed everything looks fine. if somebody have any special idea what to look for : actually we have 4 routers who are "bricked" this way - and they are not reflashed with tftp.

NeoRaider, 2016-03-25:
@viisauksena, you can check using a serial console. If the output matches https://dev.openwrt.org/ticket/21773 / https://dev.openwrt.org/ticket/21857 , this is the same bug.