freifunk-gluon/gluon #696: ntpd

Byggvir, 2016-03-21:
ntpd doesn`t update the time of the router after reboot and later. All routers are operating with different and wrong times.

'ntpd -p 0.de.pool.ntp.org' doesn`t work because:

- the name 0.de.pool.ntp.org can`t be resolved
- when using an IP-address instead the Internet is unreachable on a mesh-only router
 
```
~# nslookup 0.de.pool.ntp.org
Server:    127.0.0.1
Address 1: 127.0.0.1 localhost

nslookup: can't resolve '0.de.pool.ntp.org': Name or service not known
```
Using an IP-address works on routers connect to the Internet via WAN-port.

NeoRaider, 2016-03-21:
You need to configure a NTP server reachable via IPv6 (Gluon generally does not support IPv4 for any services except Mesh VPN).

To be able to resolve names on nodes, a IPv6 DNS server must be announced in router advertisements in the mesh, the DNS server on the WAN port is used exclusively for mesh VPN.

I've noticed that this is not covered in our documentation, I'll fix that as soon as possible.

viisauksena, 2016-03-22:
see the ntpd pool website, there you see clear statements that ONLY pool2 is for sure reachable with v6 
and this -  i think will solve your problem
here freifunk freiburg example of **site.conf** entry
```        ntp_servers = {'1.ntp.services.fffr','2.pool.ntp.org'},     -- we like to have a fallback if internal ntp fails
```

Byggvir, 2016-03-25:
'ntpd -p 2.pool.org' doesn't work, because DNS doesn't work from the command line on a mesh-only node (without wan connection). It's not a problem that you don't get an IPv6 address. Ist's the problem that you get no address.

Also IPv6 doesn't work. I tried several IPv6 addresses (eg. 'ntpd -p 2001:638:a000:1123:123::2')

Short: ntpd works only on nodes with WAN Connection.

NeoRaider, 2016-03-25:
* DNS works on mesh-only nodes if the DNS server is announced in router advertisements (actually, that has nothing to do with the WAN connection, the WAN port is never used for DNS except by fastd)
* If external IPv6 doesn't work in your community (which is possible), you need to setup an IPv6-reachable NTP server in the mesh (in Lübeck, all supernodes are also NTP servers)

viisauksena, 2016-04-18:
so this could be closed? @Byggvir @NeoRaider

NeoRaider, 2016-04-19:
I left it open as a reminder to update the docs, which I just did in 9260a8fa154b6bba559f0e46bf5597cbfaad7928.

closed 2016-04-19 by NeoRaider