freifunk-gluon/gluon #699: Mass Storage unbrick

waschmi, 2016-03-22:
I'd like to use Gluon on a Zsun Wifi Card Reader. Right now it uses the Card Read part for a Reset Button Event. That doesn't help much to enter Gluon Setup Mode if no wifi is avaible on a non-Ethernet (unless you solder it on) device. I am aware there is the possibility to flash Gluon without the need to enter the setup mode but...

1. I don't want to flash every time I'd like to change something
2. I wouldn't want to do it over insecure WiFi, even if it was avaible

So I was thinking about how handle it more gracefully, and thought my I Idea might be interesting for Gluon in general.

Basically I'd like to put a "config" on sdcard and change the gluon settings without reflash of whole firmware but that Idea could be expanded on.

Idea/ Proposal: 
1. A Demon or script will look for a Gluon-Update Folder once a new Drive/ USB-Stick/ sdcard event has been recognized. That Folder could contain a Key or Token to authenticate it's a valid and owner intended 'change' 

2. It could also contain a Firmware-File and MD5SUM for Firmware upgrade.

Basically it could be used with Router with USB or similar. I didn't read into Autoupdater yet so I'm sorry if I told an Idea that already might exist.

NeoRaider, 2016-03-25:
We don't have any interest in supporting devices without ethernet. There are enough cheap devices with ethernet...

closed 2016-03-25 by NeoRaider

pbriesch, 2016-03-25:
Hi,

even though the device has not Ethernet I would like to see it supported to some point.
The device is extremely small, has twice as much RAM and Flash Memory compared to a TL-WR841N and can be USB-powered. The following discussion in the freifunk forum shows that there is indeed interest in seeing the ZSUN stick being supported:
https://forum.freifunk.net/t/zsun-wifi-card-reader-kleinster-freifunk-router-der-welt/10923

Kind regards, Paul