freifunk-gluon/gluon #703: Changed handeling of whitespace alfred vs respondd

mmalte, 2016-03-23:
We have changed from alfred to respondd and quite some nodes are no longer shown on the map.

The Problem is additional whitespace. It was accepted in config mode:

/etc/config/gluon-node-info
```
config location
        option share_location '1'
        option latitude '50.77857023408538 '
        option longitude '6.081908941268921'
```

Alfred delivers proper data:
```
      "nodeinfo": {
        "location": {
          "latitude": 50.778570234085,
          "longitude": 6.0819089412689
        },
```

Respondd just drops the the value containing whitespace:
```
        "location": {
          "longitude": 6.081908941268921
        },
```

NeoRaider, 2016-03-23:
Yes, we've already noticed this; we had relied on LuCI's input validation, but the validation was inaccurate in this case. The validation has been improved in the master and v2016.1.x, so no new cases of this issue can occur.

How many nodes are affected on your network? We might add an upgrade script which fixes the value in the config if it is a significant problem.

yayachiken, 2016-03-23:
Approximately 40 of 1300 nodes are affected

kb-light, 2016-04-15:
Approximately 30 of 1170 nodes are affected within our community

Where to put the upgrate script? In gluon-node-info or may in an additional package?

NeoRaider, 2016-04-15:
I think *gluon-node-info* would be the appropriate place.

PR #740 (2016-04-28): gluon-node-info: add a script to fix whitespaces within the coordinates
This fixes #703 
First version, this may be improved by iterating through the listed items within _gluon-node-info_ - _location_, instead of hard coded stuff.
closed 2016-04-28 by NeoRaider