freifunk-gluon/gluon #705: Additional configs/scripts from USB storage

LittleFox94, 2016-03-25:
It would be great if gluon could search for additional/changed configuration-files and/or scripts from an USB-stick attached to the device.

Special features required by a single router could be deployed without having to build a special firmware, make the configuration available to all routers, change it manually and loose it on update ... Just put the config on a stick, attach it to the router and it's customized.

Replace USB-stick with any external storage option available to a device.

NeoRaider, 2016-03-25:
Gluon does not support USB by default (and we don't make this the default). Also, you don't provide a usecase for your proposal.

If there is any configuration that gets lost during upgrades (because our scripts overwrite it), and preserving this configuration makes sense in your opinion, please open a bug report for the specific configuration option.

Other configuration is usually coupled with additional OpenWrt packages. Preserving these during firmware upgrades usually doesn't make sense, as OpenWrt doesn't provide complete ABI/API compatiblity, so new package versions need to be installed after firmware upgrades anyways (at least when the underlying OpenWrt version is updated).

LittleFox94, 2016-03-25:
First: thank you for your fast response :)

Just being curious: why is usb not enabled by default? Just because it is not needed?

I host a Freifunk-router at home and because of a lot of technical stuff in my flat, I need the ethernet-ports to output my LAN instead of freifunk (just a simple bridge). This was possible with the old, chemnitz-specific firmware because updates were seldom and I could just change the config myself. This custom configuration now prevents gluon on my router, as updates will be more often.

@ambassador86 told me, port-configuration is already on the wish-list for the next major release, but I think this feature may come in handy for other people and use-cases too.

NeoRaider, 2016-03-25:
We try to make Gluon default configuration as similar as possible on most routers, so we don't want to  care if a device has USB or not. Some communities disagree and include USB modules by default on models with enough flash, but we don't want to rely on that.

Gluon's network configuration is very complex, which is the reason why we don't support at lot of reconfiguration there at the moment (mesh-on-lan is the most invasive exception). Of course, support for more flexible network setups is a much-requested feature, but adding a new source for configuration will not help in that regard - we rather need to reinvent the way the configuration structured and managed. The network configuration of OpenWrt stored in UCI is too low-level, to often small changes are necessary during Gluon upgrades; combining this with user configuration must be well-designed, or our upgrade script will make users unhappy when things break.

The first step in that direction will be done during this years Google Summer of Code (hopefully, if the proposal is accepted): http://lists.freifunk.net/pipermail/wlanware-freifunk.net/2016-March/003538.html . Based on the new configuration system, we'd like to create a new way to manage the network configuration; issues that have been deferred until that has been done can be found under the milestone https://github.com/freifunk-gluon/gluon/milestones/network-rewrite .

closed 2016-08-22 by rotanid