freifunk-gluon/gluon #710: posix shell issue in xtables-addons

lemoer, 2016-03-30:
I was trying to build gluon in voidlinux and got the following errors:

```
checking kernel version that we will build against... ./configure: 11842: ./configure: Bad substitution
Makefile:128: recipe for target '/home/gluon/gluon/build/ramips-rt305x/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_gluon-ramips-rt305x/linux-ramips_rt305x/xtables-addons-2.5/.configured_nnnnnnnn' failed
make[4]: *** [/home/gluon/gluon/build/ramips-rt305x/openwrt/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_gluon-ramips-rt305x/linux-ramips_rt305x/xtables-addons-2.5/.configured_nnnnnnnn] Error 2
make[4]: Leaving directory '/home/gluon/gluon/openwrt/package/network/utils/xtables-addons'
package/Makefile:191: recipe for target 'package/network/utils/xtables-addons/compile' failed
make[3]: *** [package/network/utils/xtables-addons/compile] Error 2
make[3]: Leaving directory '/home/gluon/gluon/build/ramips-rt305x/openwrt'
package/Makefile:188: recipe for target '/home/gluon/gluon/build/ramips-rt305x/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_gluon-ramips-rt305x/stamp/.package_compile' failed
make[2]: *** [/home/gluon/gluon/build/ramips-rt305x/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2_gluon-ramips-rt305x/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/gluon/gluon/build/ramips-rt305x/openwrt'
/home/gluon/gluon/Makefile:337: recipe for target 'prepare' failed
make[1]: *** [prepare] Error 2
make[1]: Leaving directory '/home/gluon/gluon/build/ramips-rt305x/openwrt'
Makefile:68: recipe for target 'all' failed
make: *** [all] Error 2
```

I figured out that they are using non posix conform shell commands in ```xtables-addons-2.5/configure.ac```.. The following fix worked for me:

``` diff
-       kmajor="${krel%%[[^0-9]]*}";
-       kmajor="$(($kmajor+0))";
-       krel="${krel:${#kmajor}}";
-       krel="${krel#.}";
-       kminor="${krel%%[[^0-9]]*}";
-       kminor="$(($kminor+0))";
-       krel="${krel:${#kminor}}";
-       krel="${krel#.}";
-       kmicro="${krel%%[[^0-9]]*}";
-       kmicro="$(($kmicro+0))";
-       krel="${krel:${#kmicro}}";
-       krel="${krel#.}";
-       kstable="${krel%%[[^0-9]]*}";
-       kstable="$(($kstable+0))";
+       kmajor=$(echo "$krel" | cut -d "." -f 1);
+       kminor=$(echo "$krel" | cut -d "." -f 2);
+       kmicro=$(echo "$krel" | cut -d "." -f 3);
+       kstable=$(echo "$krel" | cut -d "." -f 4);
+       krel=;
```

NeoRaider, 2016-03-30:
Weird. OpenWrt should always use bash for such scripts...

lemoer, 2016-03-30:
Hm. In both os (debian and voidlinux) the shebang in the generated configure script seems:

``` shell
#! /bin/sh
# Guess values for system-dependent variables and create Makefiles.
# Generated by GNU Autoconf 2.69 for xtables-addons 2.5.
```

NeoRaider, 2016-03-30:
Indeed... I'll send a bug report or a patch to the xtables-addons developers.

kpanic23, 2016-04-01:
This seems to also affect the v2016.1.3 release. I have just worked around it by linking /bin/sh to /bin/bash on my Debian build machine.

lemoer, 2016-04-01:
Well.. On a build machine this seems okay. But you should never do this on your development machine... (where you are writing code for /bin/sh)

kpanic23, 2016-04-01:
You mean like the xtables-addons developers seemingly have done? ;)
No sweat, I'm going to re-link it to /bin/dash as soon as this issue gets closed.

NeoRaider, 2016-04-01:
I wrote a patch ( http://www.spinics.net/lists/netfilter-devel/msg41458.html ), let's wait for upstream's opinion on this.

commit b46589a6 (2016-04-05): xtables-addons: build: fix configure compatiblity with POSIX shells

Fixes #710.
commit b493e1bb (2016-04-05): xtables-addons: build: fix configure compatiblity with POSIX shells

Fixes #710.
commit 4147e0e4 (2016-04-12): xtables-addons: build: fix configure compatiblity with POSIX shells

Fixes #710.
closed 2016-04-05 by NeoRaider