freifunk-gluon/gluon #716: builderror /usr/bin/env: bash: Too many levels of symbolic links

viisauksena, 2016-04-02:
uname -a 
Linux openfreiburg.de 3.2.41-042stab108.7 #1 SMP Wed Jul 15 15:49:08 MSK 2015 x86_64 GNU/Linux

as mentioned in https://github.com/freifunk-gluon/gluon/commit/d5f5ce74e8a2f7645b72bdf55ce45b86756914d6 reffered to commit 86894548c5dbaa8673dfa8e8ded3a658a63c9c6e

first buildprocess work fine, second build process failed with:
```
/usr/bin/env: bash: Too many levels of symbolic links
make: *** [manifest] Fehler 126
```
so no new firmware is  build - old is still left. Actually i recognize it because it run in nearby no time.
Removing gluon and new ```git clone github...gluon``` helps, than 1 time build is possible (which takes forever obviously)

for some people **make clean** or **make dirclean** helps, which itself shouldnt be necessary, but did not help for me (returned in error again)

until 1 week ago , there were no builderrors whatsoever

mostly i use a automatic script : https://cccfr.de/git/viisauksena/firmware/blob/support/fffrmaker
which basicly do (really cut down to essentials)
```
# update gluon and commit refference directly in own repo
git submodule update --remote gluon && git add gluon
git commit -m "update gluon" ; git push
# make FW
time 	make V=99 \
	GLUON_TARGET=ar71xx-generic \
	GLUON_BRANCH=$branch \
	GLUON_RELEASE=$release \
	GLUON_OPKG_KEY=$basedir/gluon-opkg-key \
	GLUON_OUTPUTDIR=$basedir/gluon/output/$branch \
	GLUON_IMAGEDIR=$basedir/gluon/output/$branch \
	GLUON_MODULEDIR=$basedir/gluon/output/modules	 2>&1 | tee logs/make_$branch_$release_$(date +%y%m%d_%H%M).log
# make manifest similar to this (see link above)
# sign manifest (see link above)
# make modules similar to this (see link above)
```
it doesnt seem that i do strange things to build.

NeoRaider, 2016-04-02:
Workaround:

    rm openwrt/staging_dir/host/.prereq-build openwrt/staging_dir/host/bin/bash

A proper fix will follow shortly.

viisauksena, 2016-04-02:
danke ... 
hier noch anderen denen das passiert, ich poste mal deinen workaround
https://forum.freifunk.net/t/gluon-2016-1-3-build-too-many-levels-of-symbolic-links/11573/7

closed 2016-04-02 by NeoRaider

NeoRaider, 2016-04-02:
@viisauksena, the fix can only prevent the error from newly occuring, if your build tree is already broken, you need to fix it first (the workaround should work for that).

viisauksena, 2016-04-03:
yep, this is working now - thx

commit 19f248c7 (2016-04-02): build: ensure the early path doesn't leak into later build steps

Leads to broken symlink setup by OpenWrt, usually breaking the build with
a "Too many levels of symbolic links" message.

Fixes #716.
commit 679a1996 (2016-04-02): build: ensure the early path doesn't leak into later build steps

Leads to broken symlink setup by OpenWrt, usually breaking the build with
a "Too many levels of symbolic links" message.

Fixes #716.