freifunk-gluon/gluon #721: Add support for TL-WR841N(D) Revision 11

dkessel, 2016-04-07:
Retailers are starting to sell revision 11(.1)  of the TL-WR841N(D).

It would be good to have support fro this in gluon. I found this at openwrt - DD trunk seems to have support for it:
https://wiki.openwrt.org/toh/start?dataflt%5BModel*~%5D=TL-WR841N%28D%29

NeoRaider, 2016-04-07:
OpenWrt currently supports the US version of the v11 only (TP-Link started separating firmwares between US and EU because of the new regulations). I'll get an EU test device soon, so we can add support.

belzebub40k, 2016-04-12:
Looks like all changes TP-Link made where implemented in the firmware and no significant changes where made in OpenWrt to add the support. Hopefully this is the same for the EU version.

http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=d71d620ac7d5a496bc7be16f9f1046d9c6133d14

Ranlvor, 2016-04-20:
I did an untested backport of the v11-Changes. https://github.com/Ranlvor/gluon/tree/wr841n-v11

Will pull-request as soon as I got feedback of an owner of a v11.

mmalte, 2016-04-25:
@neoraider we've got the first report of an 804v11 device today. 

Did you receive yours or should we ask for the device for testing purposes?

NeoRaider, 2016-04-25:
I don't have a 841 v11, but as there's already OpenWrt support, I'll just backport it. As I've found out, the 841 v11 does not yet have region-specific versions yet.

NeoRaider, 2016-04-25:
Added as BROKEN, please test.

closed 2016-04-25 by NeoRaider

mmalte, 2016-04-26:
I've got reports that even the 841v10 firmware is working on a 841v11

NeoRaider, 2016-04-26:
@mmalte, well, the firmwares are exactly identical, with one exception: the firmware header contains a different hardware ID, so the 841v10 firmware can't be flashed on a 841v11. Of course, first flashing OpenWrt and then flashing a "wrong" Gluon image with -F is possible and will work (but will show the device as v10).

SilversurferCH, 2016-04-27:
Hi just tested Düsseldorf firmware (http://images.ffdus.de/brokenl2tp/factory/gluon-ffdus-2016042602-exp-tp-link-tl-wr841n-nd-v11.bin). Upgrading the firmware went well. Even Configuring is working. One thing i noticed: The Power-LED is no more glowing. Actually compiling for Münster so we can test mesh.

mmalte, 2016-04-28:
I've cherry picked #8f6cc2d on top of v2016.1.4 works perfectly.

Adorfer, 2016-04-29:
I can confirm
https://karte.ffdus.de/#!v:m;n:a42bb0a9ee88 seems to work fine. 
https://map.eulenfunk.de/stats/dashboard/db/node-byid?var-nodeid=a42bb0a9ee88

mraerino, 2017-02-13:
TL-WR841N v11-EU seems not to work with gluon 2016.2.2
> Error code: 18005
> Upgrade unsuccessfully because the version of the upgraded file was incorrect. Please check the file name.

Is there something happening for this?

OpenWRT seems to somehow have support for this: https://forum.openwrt.org/viewtopic.php?pid=338779#p338779
Seems like there is only some new signature added: https://forum.openwrt.org/viewtopic.php?pid=338630#p338630

NeoRaider, 2017-02-13:
@mraerino You need to set the correct GLUON_REGION (`GLUON_REGION := eu` in *site.mk*) when building the images.

commit 8f6cc2da (2016-04-25): ar71xx-generic: add TP-LINK TL-WR841ND v11 support

Fixes #721
commit 310079cd (2016-04-29): ar71xx-generic: add TP-LINK TL-WR841ND v11 support

Fixes #721