freifunk-gluon/gluon #722: Add support for TL-WR841N(D) Revision 11

aipfel, 2016-04-07:


NeoRaider, 2016-04-07:
Support added in 5bfb06fe36348305e91cdf975c0d9d97d8cd14a2 (marked as BROKEN). Please test.

aipfel, 2016-04-21:
Okay, i tested them today and AirOS says: 'firmware check failt'. So i tried to flash OpenWRT itself but the firmware check fails too. CC 15.05 can be installed and works fine! The LAN-Port does his job aswell. What is the difference of
openwrt-15.05-ar71xx-generic-ubnt-rocket-m-xw-squashfs-factory.bin
and
openwrt-15.05.1-ar71xx-generic-ubnt-rocket-m-xw-squashfs-sysupgrade.bin
The first is working, the last doesn't ?!?

After all, i tried to install the Firmware over SSH on openwrt with -F (force) but then the rocket is broken.
Any Ideas?

by the way: original AirOS shows the same symptoms:
XM.v5.5.11.28002.150723.1344.bin 'firmware check fail'
XM.v5.5.10.24238.141001.1641.bin works

If I should try something, let me know!

NeoRaider, 2016-04-21:
@aipfel, "The first one is working, the last doesn't": If you didn't mistype, the first one is a factory image, and the second one is a sysupgrade one.

"Firmware check fail" sounds a lot like you broke u-boot's flash layout definition, which happens when an OpenWrt image with broken flash layout (with is true for 15.05.1 and everything older) is installed from certain versions of AirOS. I'm not sure if it is possible to recover from this state without a serial console, broken flash layouts are annoying...

One note: On Ubiquiti devices, it is very important to differentiate between factory and sysupgrade images. factory images are used to flash from AirOS and u-boot recovery, sysupgrade from everything OpenWrt-based.

Needing -F probably means that you tried to flash a factory image (a sysupgrade image would not need -F), in which case a bricked device is entirely expected.

aipfel, 2016-04-25:
Sorry, this was a mistype. Both were factory.

Okay, now i did the following: I updated the Rocket to 5.6.4 with the new u-boot 1.1.4-s958 and then i was able to flash CC 15.05.1, but no Gluon: Firmware check failt 

With tftp, i was able to flash the rocket back to AirOS. That's not the Problem

Do I have proceed as follows: AirOS 5.5.8 --> AirOS 5.6.4 --> CC 15.05.1 --> Gluon sysupgrade ?

NeoRaider, 2016-04-25:
@aipfel, if you don't plan to ever go back to AirOS, that's a possible way to proceed. Your flash layout definition in U-boot is probably broken, but as OpenWrt/Gluon don't care, everything should work fine.

aipfel, 2016-04-25:
okay, i try it later...

Do you stay at WCW?

NeoRaider, 2016-04-25:
Yes, I'll attend WCW.

aipfel, 2016-04-25:
okay, then I'll bring the device with me

aipfel, 2016-04-26:
"Do I have proceed as follows: AirOS 5.5.8 --> AirOS 5.6.4 --> CC 15.05.1 --> Gluon sysupgrade ?"

I test it but CC15.05.1 don't updating the rocket 
After 'upgrading' to Gluon via UI, he starts again with CC 15.05.1

Exist a way to skip the firmwarecheck from u-boot?

NeoRaider, 2016-04-26:
@aipfel, ah, of course. U-boot from AirOS 5.6.x doesn't unlock the flash, so OpenWrt CC will end up with unwriteable flash.

If you have a serial console, you should be able to flash a Gluon factory image via the "urescue" command after resetting the MTD layout (the command is something like "mtdparts default", you can find out using the "help" command.) Resetting the MTD layout is not permanent, but it is enough to make U-boot accept an image with the correct flash layout...

closed 2016-05-30 by NeoRaider