freifunk-gluon/gluon #723: Download quilt package failed

marvin, 2016-04-08:
I recommend we think about hosting the required Packages for building Gluon or creating a mirror for all the required packages because often it happens that not all required packages can be aquired.
Atm I cant build gluon because the gnome/mc-common and mm-common packages are not available.
I know this might be more related to openwrt but should we think about such a solutions as well?

```
make[4]: Leaving directory `/data1/gluon/openwrt/tools/quilt'
failed: Connection timed out.
Retrying.

--2016-04-08 08:32:43--  (try: 2)  http://ftp.gnome.org/pub/GNOME/sources/mm-common/0.9/mm-common-0.9.7.tar.xz
Connecting to ftp.gnome.org (ftp.gnome.org)|130.239.18.165|:80... failed: Connection timed out.
Connecting to ftp.gnome.org (ftp.gnome.org)|2001:6b0:e:2018::165|:80... failed: Connection timed out.
Retrying.

--2016-04-08 08:33:25--  (try: 3)  http://ftp.gnome.org/pub/GNOME/sources/mm-common/0.9/mm-common-0.9.7.tar.xz
Connecting to ftp.gnome.org (ftp.gnome.org)|130.239.18.165|:80... connected.
HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
Retrying.

--2016-04-08 08:33:48--  (try: 4)  http://ftp.gnome.org/pub/GNOME/sources/mm-common/0.9/mm-common-0.9.7.tar.xz
Connecting to ftp.gnome.org (ftp.gnome.org)|130.239.18.165|:80... failed: Connection timed out.
Connecting to ftp.gnome.org (ftp.gnome.org)|2001:6b0:e:2018::165|:80... failed: Connection timed out.
Resolving ftp.gnome.org (ftp.gnome.org)... 130.239.18.165, 2001:6b0:e:2018::165
Connecting to ftp.gnome.org (ftp.gnome.org)|130.239.18.165|:80... failed: Connection timed out.
Connecting to ftp.gnome.org (ftp.gnome.org)|2001:6b0:e:2018::165|:80... failed: Connection timed out.
Retrying.

--2016-04-08 08:35:12--  (try: 5)  http://ftp.gnome.org/pub/GNOME/sources/mm-common/0.9/mm-common-0.9.7.tar.xz
Connecting to ftp.gnome.org (ftp.gnome.org)|130.239.18.165|:80... failed: Connection timed out.
Connecting to ftp.gnome.org (ftp.gnome.org)|2001:6b0:e:2018::165|:80... failed: Connection timed out.
Giving up.

Download failed.
--2016-04-08 08:35:52--  http://mirror2.openwrt.org/sources/mm-common-0.9.7.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-04-08 08:35:53 ERROR 404: Not Found.

Download failed.
--2016-04-08 08:35:53--  http://downloads.openwrt.org/sources/mm-common-0.9.7.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-04-08 08:35:53 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.

```

mmalte, 2016-04-08:
There is quite a strong, reliable mirror server at my former dormitory:
http://ftp.halifax.rwth-aachen.de/

It's always possible to mirror free projects there for free.

marvin, 2016-04-08:
googd idea maybe we can talk to them to host gluon + all required packages there as a mirror?!?!

Adorfer, 2016-04-08:
Nur als Seitennotiz: Viele Services, die von sich behaupten, per IPv6 erreichbar zu sein, sind das nicht zuverlässig. 
Auch für apt-get, git clones aus privaten repositories etc empfielt es sich, auf ipv6 zu verzichten. (Der Strauss der Methoden dafür ist vielfältig, ich spare mir die Aufzählung)

marvin, 2016-04-08:
@Adorfer mag sein hilft uns in dem moment aber nicht wirklich weiter oder?

Adorfer, 2016-04-08:
Naja, im konkreten Fall bräuchte man wohl wirklich einen "stale-mirror", denn das besagte
Download failed.
--2016-04-08 08:35:52--  http://mirror2.openwrt.org/sources/mm-common-0.9.7.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-04-08 08:35:53 ERROR 404: Not Found.

tritt ja nur auf, da das Paket inzwischen mm-common-0.9.8.tar.xz heisst

marvin, 2016-04-08:
@Adorfer was wäre den deine idee dazu das zu lösen?

NeoRaider, 2016-04-08:
ftp.gnome.org is working again.

OpenWrt trying to download the file from sources.openwrt.org and mirror2.openwrt.org is completely normal; it is the hardcoded fallback. OpenWrt usually won't mirror files here when the primary mirror is working fine, so getting a 404 for these domains is completely normal.

closed 2016-04-08 by NeoRaider