freifunk-gluon/gluon #724: Preparing Gluon for Babel

tcatm, 2016-04-08:
Basically, fix all implicit deps on batman-adv in various packages:

- [ ] `gluon-radvd` refers to `bat0` here https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-radvd/files/lib/gluon/ebtables/300-radv-input-output
- [ ] `gluon-radvd` must run on `local-node` instead of `br-client` (for the next-node-mac to become the gateway)

christf, 2016-04-14:
the following packages also do contain references to "bat"

gluon-mesh-vpn-fastd
gluon-respondd (hotplug-script)
gluon-status-page
gluon-status-page-api

christf, 2016-04-23:
gluon-mesh-vpn should add the mesh-vpn interface to /lib/gluon/core/dynamic/mesh_interfaces, one interface per line.

tcatm, 2016-05-03:
- [ ] network.local_node_route6 must be removed

christf, 2016-11-19:
most of this is done in #934 
the following is still open
- statuspage / statuspage-api
- gluon-respondd

christf, 2016-11-25:
gluon- respondd is mesh-protocol agnostic with #946

christf, 2018-01-07:
I do not think we need  respondd to notice if interfaces change for a new mesh protocol. Sure, it should be done but is not necessary for babel.

christf, 2019-09-25:
the babel-speciic parts have been implemented. Creating a new issue for respondd

closed 2019-09-25 by christf