freifunk-gluon/gluon #727: bla - bridge loop avoidance

viisauksena, 2016-04-18:
somebody tried this - we wonder if we can use all our servers 
so that they claim the router and we disable cross traffic and some other stuff
a first very simple test seems to look good. 
So one server claimed very fast all ra - we will run a bigger test with all Servers (who have exit ) and see if the claiming level out better. 
https://www.open-mesh.org/projects/batman-adv/wiki/Bridge-loop-avoidance-II
this is what we want 
https://www.open-mesh.org/attachments/download/142/mesh_LAN_mesh.dia.png

NeoRaider, 2016-04-19:
I'm not exactly sure what this ticket is about. It seems this is about backbone design, which should rather be discussed on the Gluon ML and not in the issue tracker.

jjx-, 2016-04-19:
Are you sure you activated bla on more than one Server?

viisauksena, 2016-04-21:
we activated it on all our servers, but not on the nodes - it seems to work fine

viisauksena, 2016-05-03:
so no negative influences so far , by 300 nodes and 8 Supernodes/Gateways for 2 Weeks, 
dont know how this issue could make progress, or whatever .. 
we enabled this for the servers , but not for the nodes so far.
i wonder how their could be progress here - or just to close it  - since is no more an issue whatsoever

Adorfer, 2016-05-04:
> so no negative influences so far

Would you notice dropped packets due to false positives?
I heared about strange effects with bla on batman v14, resulting in users turning it off again... but that is years in the past. 
So nobody seems to remember what happened exactly and how it was traced down.

viisauksena, 2016-05-08:
after talking some time with linus from batman-adv project it became clear that we do not need it, but that this is suggested to be default on
because we dont need it in our setting, there are obviously no scenarios where this packetloss could happen - so that is something i cannot tell anything

T-X, 2016-05-09:
Just to clarify to others: The setup @viisauksena described to me did not seem to need any BLA. There seems to have been a small confusion regarding multiple paths below vs. "on top" of a bat0 interface. Below, that's where the routing protocol does its job to find a loop-free route, even if there are multiple paths available.

On top of a bat0, if you somehow connect bat0's of different nodes via bridges, then you need BLA. In @viisauksena scenario there was no such bridging involved.

@viisauksena: Just one more note for clarification: If you were having the client network on your typical TP-Link Freifunk node on the LAN ports (default in Gluon). And connected two nodes via a LAN cable between these LAN ports. Then you'd need to activate BLA on these two nodes to avoid a loop and packet storm.

closed 2016-05-14 by T-X

rubo77, 2019-01-02:
@T-X : if you need to activate it if you connect two nodes via LAN, then this option would be useful to prevent problems in networks, where cleaningwomen are causing havoc.

can you activate this per default in site.con somehow? Or would we have to modify gluon for this to be active on all nodes?

Would it be a problem to activate this on all nodes? what about batman-14 with bla on?

There is nothing in the gluon doku, maybe we should add a section there?

rubo77, 2019-01-03:
@neoraider told me: 

> IIRC, BLA is always enabled for compat 15, and always disabled for compat 14