freifunk-gluon/gluon #728: special banner with advanced ssh login info and miniscripts

viisauksena, 2016-04-18:
interest in obtaining 
1. much more info on login via ssh
2. some really basic shell scripts which make config easy 
if so : i could build some generic ones .. maybe called gluon-ssh-helpers
```
# helper scripts to turn on or off, or set values 
# selfexplaining, all with usage
/usr/bin$ ls
autoupdater-status  branch  clientwifi  contact  fastd-status  geoguess  help  hop_penalty  location  lon lat alt MoL  MoW  nodeinfo  nodename  sshkeys  status  v4up
```
```
# example ssh-login 
#
########################################
checking v4 - empty if v4 available 
you may want IPv4, try :   udhcpc -i br-client -q -S -n
########################################
### IP :            inet6 addr: 2001:bf7:2810:ff42:c66e:XXXX:XXXXXXXXX/64 Scope:Global
### from ssh : fe80::224:d7ff:XXXXXXX%br-client 40511 fe80::c66e:1fff:XXXXXXX%br-client 22
### known sshkeys : fuzzle birger mar fabzgy@fabzgy wuschl fuzzle - active : 1
### active sshkeys : fuzzle (1 of 6)
### uptime/load :   01:51:40 up 9 days, 10:16,  load average: 0.07, 0.17, 0.20
########################################
### pubkey :  dbddc1b76XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX85645
### firmware :  v2016.1.3   branch :  stable   autoupdater : 0
### hardware :  TP-Link TL-WR841N/ND v9
########################################
### MoL:'1'   MoW:'0'   fastd:'1' clientwifi:1
### Location:  '0' lat: lon: alt:
### contact :  'kaulquappen supäää'
########################################
### connected to this node  :    0
### number of nodes (inkl SN) :  307
### number of total clients :    541
### number of GW :  8  - connected to =>        Master-SN4 (234) ea:e1:28:f9:09:08 [     ibss0]:  79 - 32MBit/32MBit
########################################
### /sys/class/net/bat0/mesh/hop_penalty :    33
### batctl gw :      client (selection class: 20)
####logread#############################
Tue Apr 19 01:51:40 2016 daemon.info fastd[1465]: resolving host `sn10.freiburg.freifunk.net' for peer <mesh_vpn_backbone_peer_sn10>...
Tue Apr 19 01:51:40 2016 daemon.info fastd[1465]: resolving host `sn10.freiburg.freifunk.net' failed: Name or service not known
Tue Apr 19 01:51:41 2016 daemon.info fastd[1465]: resolving host `sn8.freiburg.freifunk.net' for peer <mesh_vpn_backbone_peer_sn8>...
########################################
```

NeoRaider, 2016-04-19:
Having a more useful banner would be nice, but there's also a downside: network connectivity to mesh nodes is sometimes flaky, and in such a case any surplus data on login is a burden. Mostly an admin logging into a node is interested in one specific piece of information, a banner would either be very long or most likely useless.

Allowing high-level configuration on the shell is a completely separate issue (please open one ticket per issue!) This should be solved by refactoring the config scripts of the expert mode into a Lua library and using the same implementation for both the expert mode and on the shell. Doing this now  is probably not a good idea, as we plan to rework the way Gluon handles configuration (by not using UCI as the main config storage, but generating UCI from another config system; this will be part of my Google Summer of Code project, if my proposal is accepted.)

viisauksena, 2016-04-20:
if you had bad connection you can do maintainance "blind" .. 
**ssh bla@foo "command"** should do the trick .. then you dont get the **/etc/profile** data - but its also possible to make a BIG note to use buildin **helper** - which will then do the job ... the helper scripts are very ver useful - for me and for many people who now the shell only a little. This fill  the missing config possibilities while nodes are runing.

belzebub40k, 2016-04-20:
Maybe it would be an idea to minimize the banner to something like: `type 'help' for more information`

And implement a helper script named `help` which shows some stats about the node and hits for other useful commands.

viisauksena, 2016-07-14:
i renew the question .. would this be merged, and prefferable where ... 
i have done 1 bit package .. which easily could be split in two,
one miniscript package and based on that one info package .. for much more info. 
(basicaly like above example) - i could clean it up and make a push request.

NeoRaider, 2016-07-27:
Personally, I don't like the banner very much, so I'd like to get some more feedback on this feature from other users. @belzebub40k's suggestion sounds like a good idea.

As written before, it is a bad time to add configuration helper scripts, as they would need to be rewritten soon.

viisauksena, 2016-07-27:
i agree fully ... all the miniscripts are at least nothing more than "clever" utilized commands and could sit in a special help or nodeinfo file - which make nodeinfo a bit trickier ... 
as i have already build some for freiburg long ago - it should be easy done on a rainy afternoon - and also easy rewriteable when things change in the background. (because waiting for undefined is boring - no offense meant ;)

SimJoSt, 2016-12-12:
I would like to comment on the banner feature without cluttering this ticket. @viisauksena can you please split the two suggestions here in two separate tickets?
I also wouldn't have a problem with discussing the banner in #901, as it captures my intentions already.
Some clarity needed.

viisauksena, 2016-12-12:
mhm, yes it would be better to have these miniscripts and advanced info seperated.
(but this is also not soo easy because they easy depend on each other - and also there will be a tendency to have this not pinned to many possible variants , like mesh (babel, ibss, 80211s), and uplink (wireguard, fastd, l2tp) and the existance of LAN/WAN , 2g/5g , whatever fancy/default package and so on ..  so it may sound straight forward but it isnt necessarily)

and this issue is pretty on hold, at least until Gluon goes LEDE ...
you saw the proposal from @rubo77 in #901 ... 
```
lua -e 'print(require("platform_info").get_board_name())'
lua -e 'print(require("platform_info").get_model())'
cat /lib/gluon/gluon-version /lib/gluon/release
```

viisauksena, 2017-04-10:
the script is still available on 
https://github.com/viisauksena/gluon-banner
(and so can be added via modules and entry in site.mk - it is slightly freiburg based)
but i dont see that this ever is going to come in gluon - maybe in future time when lede is settled and there is some time

closed 2017-04-10 by viisauksena