freifunk-gluon/gluon #729: Add Support for TP-Link WBS210 & WBS510

mmalte, 2016-04-19:
We've got Hands on a TP-LINK WBS210 and a WBS510:

Qualcomm Atheros Enterprise AR9350B 560MHz CPU, MIPS 74Kc
64MB DDR2 RAM, 8MB Flash
1 10/100Mbps Shielded Ethernet Port (LAN0/POE)
1 10/100Mbps Shielded Ethernet Port (LAN1)
2 RP-SMA

see:
http://www.tp-link.com/en/products/details/WBS210.html#specifications

Seems like there is no openwrt support yet, but we'll like to change that.

What information do we need to provide.

NeoRaider, 2016-04-19:
Regarding software, these devices should be identical to the CPE210/510, they also use the "PharOS" stock firmware.

Adding the additional strings "WBS210(TP-LINK|UN|N300-2):1.0" and "WBS510(TP-LINK|UN|N300-5):1.0" (taken from PharOS image) to *cpe510_support_list* in *tools/firmware-utils/src/tplink-safeloader.c* might be enough to support them.

mmalte, 2016-06-20:
Are you sure about the 1.0? 
[image]

We've got hands on a 1.2

I'm not sure about the necessary changes in targets/ar71xx-generic/profiles.mk as well. I've added:

```
$(eval $(call GluonModelAlias,CPE510,tp-link-cpe510-v1.0,tp-link-wbs510-v1.2))
$(eval $(call GluonModelAlias,CPE510,tp-link-cpe510-v1.0,tp-link-wbs210-v1.2))
```

Do I need to change the GluonModel line as well?

NeoRaider, 2016-06-21:
First, a patch to the image generator is necessary. I can have a look at the stock images and adjust the image generator accordingly.

mmalte, 2016-06-21:
The Firmware on at the TP-Link Homepage haven't changed since 15/12/15

http://www.tp-link.de/download/WBS210.html#Firmware

NeoRaider, 2016-06-21:
This is the support list in the current firmware:
```
CPE510(TP-LINK|UN|N300-5):1.0
CPE210(TP-LINK|UN|N300-2):1.0
WBS510(TP-LINK|UN|N300-5):1.0
WBS210(TP-LINK|UN|N300-2):1.0
CPE510(TP-LINK|UN|N300-5):1.1
CPE520(TP-LINK|UN|N300-5):1.1
CPE210(TP-LINK|UN|N300-2):1.1
CPE220(TP-LINK|UN|N300-2):1.1
WBS510(TP-LINK|UN|N300-5):1.20
WBS210(TP-LINK|UN|N300-2):1.20
CPE510(TP-LINK|US|N300-5):1.1
CPE520(TP-LINK|US|N300-5):1.1
CPE210(TP-LINK|US|N300-2):1.1
CPE220(TP-LINK|US|N300-2):1.1
WBS510(TP-LINK|US|N300-5):1.20
WBS210(TP-LINK|US|N300-2):1.20
CPE510(TP-LINK|EU|N300-5):1.1
CPE520(TP-LINK|EU|N300-5):1.1
CPE210(TP-LINK|EU|N300-2):1.1
CPE220(TP-LINK|EU|N300-2):1.1
WBS510(TP-LINK|EU|N300-5):1.20
WBS210(TP-LINK|EU|N300-2):1.20
CPE210(TP-LINK|EU|N300-2):2.0
CPE210(TP-LINK|UN|N300-2):2.0
TL-CPE210(TP-LINK|CN|N300-2):1.0
TL-CPE510(TP-LINK|CN|N300-5):1.0
TL-BS210(TP-LINK|CN|N300-2):1.0
TL-BS510(TP-LINK|CN|N300-5):1.0
TL-BS210(TP-LINK|CN|N300-2):1.20
TL-BS510(TP-LINK|CN|N300-5):1.20
```

mmalte, 2016-06-30:
Thank you for looking up the values. For the first try to build the firmware I had added:
WBS210(TP-LINK|UN|N300-2):1.2
Now it is workinkg with your string:
WBS510(TP-LINK|UN|N300-5):1.20

https://map.aachen.freifunk.net/#!v:m;n:ec086b4be464

The only issue I see, is that the Harware is detected as:
TP-LINK CPE210/220/510/520

Here is the dmes of the device:
https://gist.github.com/mmalte/a56249d8be4aa6c309c24dbf59b7d394

Does this dump contain the information needed to fix the Model String or can we provide something else?

Tarnatos, 2016-07-17:
Are there any current plans to bring the support upstream?

NeoRaider, 2016-07-27:
I will probably add the WBS support in upstream LEDE soon.

NeoRaider, 2016-07-27:
@mmalte, could you provide a dump of your `product-info` MTD partition (as shown in `/proc/mtd`)?

kb-light, 2016-08-05:
@NeoRaider: I have a dump of product-info in case you need more informations than already provided in #844

kb-light, 2017-02-05:
support was added by 99471f8, so this issue can be closed.

kpanic23, 2017-02-06:
Last week I managed to get hold of a WBS210. I have tested running Gluon on it. But although I had connected two 18dBi Yagi Antennas, reception was worse than with a WR841N v9. Also, Transmit Power could not be set higher than 8dBm. So I guess there still is a bit of work to be done.
But at least it runs with Gluon.
[image]

rotanid, 2017-02-07:
@kb-light you're right.
@kpanic23 please open a new issue for problems with the device, it was actually added as BROKEN because of missing tests

closed 2017-02-07 by rotanid